pythonscad/pythonscad #3: Python unusable, Debian 12, build from source

davidmcnabnz, 2025-03-06:
**Describe the bug**
I built PythonSCAD today from fresh git pull, and followed the build instructions carefully.
OS is Debian 12, x86-64.

**To Reproduce**
1. Follow the build instructions for current Linux, from the first `git clone`, to running the recommended help scripts, the cmake command, then the make command.
2. In the `build` directory, open the app with `./pythonscad`
3. When the app shows its open/create dialog, click the green `Python` button
4. When the app main window opens, click `File` -> `Examples` -> `Python` -> `simple.py`
5. Notice that the file code appears in the left code pane
6. Notice that the console pane displays the message:
```
Loaded design '/home/david/Nextcloud/3d/apps/pythonscad/examples/Python/simple.py'.
[ERROR: Parser error: syntax error in file simple.py, line 1](https://github.com/pythonscad/pythonscad/1,/home/david/Nextcloud/3d/apps/pythonscad/examples/Python/simple.py)
Execution aborted
```
It appears that the file is opened  but the app is feeding it to the OpenSCAD interpreter, not the Python interpreter.

Also, the `Open` system modal dialog's "file type" dropdown (above the Open/Cancel buttons) only shows `OpenSCAD Designs`, and doesn't offer the `Python *.py` option.

**Expected behavior**
When open the same file in the 2024-12-19 AppImage build, the file executes and previews the cube normally, as per this console pane output:
```
Loaded design '/home/david/scad/simple.py'.
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 1
Geometry cache size in bytes: 896
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized tree has 1 elements!
Compile and preview finished.
Total rendering time: 0:00:00.084
```

**Code reproducing the issue**
Not relevant.

**Screenshots**
Above explanations should suffice.

**Environment and Version info (please complete the following information):**
OS: Debian 12
System: Standard Intel x86-64 based laptop
OpenSCAD version: 

 - OS: [e.g. Windows 10, macOS 12 Monterey, Ubuntu 22.04]
 - System: [e.g. Intel PC 64-bit, Raspberry PI 4]
 - OpenSCAD Version [e.g. 2021.01, 2023.04.16] and where it was installed from

**Library & Graphics card information**
Here it is:
```
OpenSCAD Version: 2025.03.07
System information: Linux 6.1.0-28-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.119-1 (2024-11-22) x86_64 Debian GNU/Linux 12 (bookworm) 12 CPUs 46.72 GB RAM
User Agent: OpenSCAD/2025.03.07 (Linux x86_64; Debian GNU/Linux 12 (bookworm))
Compiler: GCC "12.2.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_74
Eigen version: 3.4.0
CGAL version, kernels: 5.5.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Clipper2 version: 1.4.0
Manifold version: 3.0.1
Qt version: 5.15.8
QScintilla version: 2.13.3
InputDrivers: 
GLib version: 2.74.6
[... 417 lines cut ...]
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_shader_interlock
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
```

**Additional context**
The December 2024 AppImage works mostly ok. The January 2025 AppImage complains with "Python Not found". 
My Debian 12 installation is running Python 3.11.2

gsohler, 2025-03-06:
Hi David,

Thank you for the excellent Bug report!

I believe python was not enabled during your build.

After cmake step please carefully watch the output.
It should display something like:

-- libzip: 1.9.2
-- Freetype: 2.12.1
-- LibXml2: 2.10.4
-- Python enabled, using Nettle 
-- Libs  /usr/lib64/libpython3.12.so 
-- include  /usr/include/python3.12 
-- Python enabled
-- HidAPI: 0.13.1
-- SpNav: disabled
-- CAIRO_INCLUDEDIR: /usr/include

Messages which should not appear are:

Nettle not found, disabling python support.
Python not found, disabling python support.

in either case make sure that you got nettle-dev or python-dev installed in your system

let me know!

davidmcnabnz, 2025-03-07:
Thanks gsohler,

I did already have `python3-dev` and `libpython3-dev` installed, but no nettle.

I've deleted/recreated `build`, installed `nettle-bin` and `nettle-dev`, repeated the cmake and started make, this time capturing all outputs to log files. I'll let you know soon how this goes.

D

davidmcnabnz, 2025-03-07:
I'm happy to report that following the build recipe after installing nettle-bin/nettle-dev seems to have resolved the issue. 

I guess this bug can be easily resolved by adding `python3-dev` and `nettle-dev` to the relevant prerequisites scripts.

Thanks for your help
Cheers
David

gsohler, 2025-03-07:
Hi David,

Thank you for your positive feedback.
Right now I have updated the uni-get-dependencies.sh to install the
important dependencies in Debian
Thank you for the correct package names BTW!
Also I made a small tweak yesterday to hide the "Python" button in the
Launcher in case it get compiled without Python.

However I could not reproduce how you got the Python examples into the
source code editor.
When I tested, all of them just turned into an empty "Untitled.scad" script
- strange.

In any case I am happy you got your issue  solved so quickly
Feel free to let us know, how it goes and if you could imagine "missing"
features.

Nice greetings from the other side of the world - Austria

closed 2025-03-17 by gsohler