pythonscad/pythonscad #4: Facilitate script-directory-relative accesses

davidmcnabnz, 2025-03-15:
**Is your feature request related to a problem? Please describe.**
When a script's execution is triggered by Load/Save, there is nothing in the namespace of the script, or the foundational `openscad` module, which lets a script detect what directory it resides in.

In my other Python programs, I do this discovery to support importing and asset loads relative to the script's location with boilerplates such as:
```
import sys, os
THIS_DIR = os.path.dirname(os.path.abspath(__file__))
```
Sometimes, the script is in a first-level subdirectory of a particular source tree, so I follow this up with:
```
ROOT_DIR = os.path.dirname(THIS_DIR)
if ROOT_DIR not in sys.path: sys.path.insert(0, ROOT_DIR)
```

However, in the PyOpenSCAD runtime environment, the all-important `__file__` variable is suppressed from the namespace, and `os.getcwd()` returns the main PyOpenSCAD library path eg `~/.local/share.OpenSCAD/libraries`

**Describe the solution you'd like**
I'd be happy with anything that lets the running script to determine either its absolute pathname. Possible options include:
1. changing `cwd` to the directory in which the script is executing, so that `os.getcwd()` can be used for determining the absolute directory, and/or
2. exposing a small convenience function into the `openscad` module which returns either the absolute pathname, or the directory name, of the script which is running, eg:
```
def getscriptdir() -> str:
    # magic happens here
    # ...
    return script_dir
```

**Describe alternatives you've considered**
My only workaround is assigning a variable at the top of every script to the directory in which the script is located. But this breaks the script if I move it, or any of its ancestor directories, so it adds maintenance burden.

gsohler, 2025-03-17:
This is a good addition!
I can create a small function to get the path of the stored script on disc.

gsohler, 2025-03-17:
fixed via 70d8cb5f5..6ac0a7462  master -> master

[image]

closed 2025-05-18 by gsohler