pythonscad/pythonscad #5: Add "IPython" Integration Option

davidmcnabnz, 2025-03-16:
**Is your feature request related to a problem? Please describe.**
In trial-and-error loops, it feels limiting to have to Save the file before it gets executed. It would be great to have some kind of interactive console such as IPython

**Describe the solution you'd like**
It would be awesome if there was some kind of way to type statements into an IPython console and see the statements take effect immediately. There are a couple of ways to do this:
1. Implementing a new Window type within the app, where at the end of executing the current Python file, an IPython REPL prompt appears, so the user can type commands such as: `part2.scale([1.5, 1,2, 1]).show()` and see the effect immediately. Or,
2. Providing a way for an external IPython console to access the PyOpenSCAD environment. For example, a client/server framework where an external program can do something like:
```
# load IPython engine
from IPython import embed

import openscad_client as C
# the act of importing causes a client connection to occur to a running instance of PyOpenSCAD

# this enters an IPython REPL, which has `C` in its namespace
embed()
```
then user, within IPython REPL, could do something like:
```
In [1]: c = C.cylinder(r1=40, r2=33, h=40, fn=100)
In [2]: c.show()
... c displays in PyOpenSCAD instance's viewer ...
In [3]: 
```

**Describe alternatives you've considered**
I've considered implementing a server within a running PyOpenSCAD script, and exposing its namespace to a client class via proxy objects, but I'm not sure if this would be workable. It seems PyOpenSCAD's Python setup requires the ability to run its script to completion, otherwise it becomes unresoponsive.

gsohler, 2025-03-17:
Hi David, I love your intent!
however, I am yet still quite unexperienced with IPython module.
Instead, few month back I have started a quite comparable approach.
I was able to use pythonscad as a Kernel in Jupyer.
The branch for that is pythonscad/jupyter and i have just synced it to the latest master today
It involves compiling libxeus manually which provides a jupyter protocol server
Please use ENABLE_JUPYTER for that.
The only reason, why its not yet merged to master is because i did not manage to see 3d viewer cells in jupyter

If you have some extra experince there, i'd love to get some collaborative input.

PS: please elaborate little bit on "PythonSCAD instance' because i am not exactly sure where pythonscad is living in your description.

davidmcnabnz, 2025-03-17:
I think I may have failed to explain properly.
By "PythonSCAD instance", I'm referring to an instance of the PyOpenSCAD app, open, with its window showing the normal UI components including the viewer.

It occurred to me that it might be possible to embed a server into PyOpenSCAD, possibly HTTP-based API (JSON-RPC or REST), listening on a configured address (default `127.0.0.1`, but user could set to a LAN IP such as `192.168.11.21`).

Then, a client program could connect in. There could be a client library which:
1. initiates a connection to the running server
2. creates an object, which acts as a "proxy" to the namespace of the currently-loaded python script.

For example, if the PyOpenSCAD UI had loaded a script with:
```
from openscad import *

rShaft = 4
thickness = 1

rOpening = 20
hShaft = 20
hOpening = 20
res = 100

shaftOuter = cylinder(r=rShaft+thickness, h=hShaft, center=True, fn=res)
shaftInner = cylinder(r=rShaft, h=hShaft+0.01, center=True, fn=res)
shaft = shaftOuter - shaftInner

openingLift = [0, 0, hOpening/2 + hShaft/2]
openingOuter = cylinder(r1=rShaft+thickness, r2=rOpening+thickness, h=hOpening, center=True, fn=res)
openingInner = cylinder(r1=rShaft, r2=rOpening, h=hOpening+0.01, center=True, fn=res)
opening = openingOuter - openingInner + openingLift

model = shaft | opening
model.show()
```
then with the 'client object' (call it `O`), one could execute:
```
>>> m = O.model
>>> m1 = m * [1, 1, 1.5]
>>> m1.show()
```
Then see in the window's viewer a vertically-stretched version of the funnel.

For this to happen, the client class would have a `__getattr()__` which returns a 'proxy object' for each fetched method, then executes an API call to the server. For instance, `somePart + [0, 0, 5]` would execute `__getattr__('__add__')`. Then it would get a proxy object `p`, and then execute `p.call('__add__', [0, 0, 5])`, and the `.call()` would return another proxy object.

gsohler, 2025-03-17:
Hi David,

I think I understand now
You want to add an option to remote-control pythonscad from outside and you
want it to execute python code you send.
Ultimately you want to do baby-steps and slowly extending your model by
try-and-fail

And yes, already have similar, but not equal solutions to that

* --ipython method

[image: image.png]
disadvantage: need external tool to do the graphics

*) bare python method
 start bare python,  import pythonscad module (which sits in site-packages)
not working "right now" but before

disadvantage: no graphical output either

*) jupyer lab, its a web based version of ipython ,
you need to setup a small jupyter kernel file

[image: image.png]

disadvantage: did not yet manage to insert 3d viewers into the browser
using pythreejs

*) your proposal to start a server. Yes its possible we have python locks
which make sure thread safety

no disadvantage: you know how to create/setup the client ?

Whatever version you prefer. I like pythonscad to become more "interactive"
,i am happy to support you

let me know

davidmcnabnz, 2025-03-18:
I haven't learned Jupyter, so I'd have to put some time into it before I could comment on that as an option.

I can share that I can tighten my trial/error loops through lots of print() statements, as well as a way to "stop everything, render something and stop:
```
class RenderQuit(Exception):
    """
    for quitting out for a test render
    """
    def __init__(self, model, comment=None):
        super().__init__()
        if comment:
            print(f"*** RenderQuit: {comment} *********")
        else:
            print("*** RenderQuit ***********")
        if comment:
            print(comment)
        try:
            model.show()
        except Exception as e:
            print("failed to render:")
            traceback.print_exc()
```

This allows me to sprinkle checkpoints through the code, eg:
```
def create():
    rShaft = 4
    thickness = 1

    rOpening = 20
    hShaft = 20
    hOpening = 20
    res = 100

    shaftOuter = cylinder(r=rShaft+thickness, h=hShaft, center=True, fn=res)
    shaftInner = cylinder(r=rShaft, h=hShaft+0.01, center=True, fn=res)
    shaft = shaftOuter - shaftInner

    raise RenderQuit(shaft, "just the shaft")

    openingLift = [0, 0, hOpening/2 + hShaft/2]
    openingOuter = cylinder(r1=rShaft+thickness, r2=rOpening+thickness, h=hOpening, center=True, fn=res)
    openingInner = cylinder(r1=rShaft, r2=rOpening, h=hOpening+0.01, center=True, fn=res)
    opening = openingOuter - openingInner + openingLift

    model = shaft | opening
    return model

try:
    model = create()
    model.show()
except RenderQuit as e:
    pass
```

The reason for exception-based interruption is to cover cases where the code is several functions deep.

But back to the point -- I can understand if it's non-trivial to either add an extra window type with a Python REPL console, or support an external Python client.

gsohler, 2025-03-18:
the exception strategy is actually  nice thing to display sub-nodes.
so far i have had some show() statement in the end of the files and i just
had enabled one at a time.
will thing about, if the language can offer general help here.

as for jupyter, i suggest that you 1st try it to do some calculations like
* install it with your package manager
* jupyter notebook or juptyer lab
*  hit 5+5 <shift+Enter>
* watch youtube videos about it
* get libxeus compiled(had some dependencies)
* compile libxeus wirth pythonscad
*  have a file in /usr/local/share/jupyter/kernels/pythonscad/kernel.json
with:
{
 "argv": [
  "/home/gsohler/git/pythonscad/build/openscad",
  "--jupyter",
  "{connection_file}"
 ],
 "display_name": "PythonSCAD",
 "language": "python",
 "metadata": {
  "debugger": true
 }
}
* select pythonscad as a kernel when opening a new notebook
All this should help with a speed-start.
I also have some code to display 3d stuff in jupyter, but i believe its
enough for today
(Jupyter also uses IWidget class i believe)

haha, have a nice evening

closed 2025-05-18 by gsohler