pythonscad/pythonscad #8: Error starting the application

gummist, 2025-03-29:
**Describe the bug**
PythonSCAD dies when starting with the message:
mimalloc: error: buffer overflow in heap block 0x020005F1FE80 of size 21: write after 21 bytes

**To Reproduce**
Happens everytime when attempting to run PythonSCAD

**Environment and Version info (please complete the following information):**
 - OS: Linux Mint 22.1
 - System: Intel PC 64-bit
 - OpenSCAD Version: latest version from github (29/3/2025)
 - Python version: 3.12.3
 - Cmake version: 3.28.3
 - Qt version: 682

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

Cannot run openscad to verify this

**Additional context**
Last time I updated, compiled and used PythonSCAD was about a week or so. I used the same equipment and development software without any problems.
But today I updated (git) my local sources and have this error.

No errors compiling the application.
Running the application without passing any args.

The error arises before any windows appears, after calling 'show()' in line 946 of gui/MainWindow.cc.

I compiled the application in Debug mode and I include the gdb backtrace:
.
. (debugging line by line in gui/MainWindow.cc)
.
946	  show();
(gdb) 
mimalloc: error: buffer overflow in heap block 0x020005F1FE80 of size 21: write after 21 bytes

Thread 1 "pythonscad" received signal SIGABRT, Aborted.
__pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>)
    at ./nptl/pthread_kill.c:44

(gdb) bt

#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>)
    at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=<optimized out>, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff224527e in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff22288ff in __GI_abort () at ./stdlib/abort.c:79
#5  0x000055555655064d in mi_error_default (err=14)
    at /local/proyectos/pythonscad/submodules/mimalloc/src/options.c:548
#6  0x0000555556550790 in _mi_error_message
    (err=14, fmt=0x555556c12028 "buffer overflow in heap block %p of size %zu: write after %zu bytes\n") at /local/proyectos/pythonscad/submodules/mimalloc/src/options.c:579
#7  0x0000555556546936 in mi_check_padding (page=0x20005c01300, block=0x20005f1fe80)
    at /local/proyectos/pythonscad/submodules/mimalloc/src/free.c:501
#8  0x0000555556545576 in mi_free_block_local
    (page=0x20005c01300, block=0x20005f1fe80, track_stats=true, check_full=false)
    at /local/proyectos/pythonscad/submodules/mimalloc/src/free.c:35
#9  0x0000555556545aa6 in mi_free (p=0x20005f1fe80)
    at /local/proyectos/pythonscad/submodules/mimalloc/src/free.c:155
#10 0x00007ffff7fa5f50 in ??? () at /lib/x86_64-linux-gnu/libxcb-cursor.so.0
#11 0x00007ffff7fa66f1 in xcb_cursor_load_cursor () at /lib/x86_64-linux-gnu/libxcb-cursor.so.0
#12 0x00007fffec6e9a51 in QXcbCursor::createFontCursor
    (this=this@entry=0x200000d30c0, cshape=cshape@entry=0)
    at /home/qt/work/qt/qtbase/src/plugins/platforms/xcb/qxcbcursor.cpp:501
#13 0x00007fffec6ea183 in QXcbCursor::changeCursor
    (window=0x20005f8a200, cursor=<optimized out>, this=0x200000d30c0)
    at /home/qt/work/qt/qtbase/src/plugins/platforms/xcb/qxcbcursor.cpp:335
#14 QXcbCursor::changeCursor (this=0x200000d30c0, cursor=<optimized out>, window=0x20005f8a200)
    at /home/qt/work/qt/qtbase/src/plugins/platforms/xcb/qxcbcursor.cpp:314
#15 0x00007ffff343fe1d in QWindowPrivate::applyCursor (this=this@entry=0x20005e04740)
--Type <RET> for more, q to quit, c to continue without paging--
    at /home/qt/work/qt/qtbase/src/gui/kernel/qwindow.cpp:3193
#16 0x00007ffff344634b in QWindowPrivate::setCursor
    (this=0x20005e04740, newCursor=newCursor@entry=0x7fffffffb990)
    at /home/qt/work/qt/qtbase/src/gui/kernel/qwindow.cpp:3174
#17 0x00007ffff344643d in QWindow::setCursor (this=<optimized out>, cursor=...)
    at /home/qt/work/qt/qtbase/src/gui/kernel/qwindow.cpp:3134
#18 0x00007ffff3fd5dff in applyCursor (c=..., w=0x2000024ac00)
    at /home/qt/work/qt/qtbase/src/widgets/kernel/qwidget.cpp:5030
#19 qt_qpa_set_cursor (w=w@entry=0x2000024ac00, force=force@entry=false)
    at /home/qt/work/qt/qtbase/src/widgets/kernel/qwidget.cpp:5068
#20 0x00007ffff3fe35f2 in qt_qpa_set_cursor (force=false, w=0x2000024ac00)
    at /home/qt/work/qt/qtbase/src/widgets/kernel/qwidget.cpp:5041
#21 QWidgetPrivate::show_sys (this=this@entry=0x200028d8480)
    at /home/qt/work/qt/qtbase/src/widgets/kernel/qwidget.cpp:8148
#22 0x00007ffff3fec45f in QWidgetPrivate::show_helper (this=this@entry=0x200028d8480)
    at /home/qt/work/qt/qtbase/src/widgets/kernel/qwidget.cpp:8074
#23 0x00007ffff3feff8b in QWidgetPrivate::setVisible (this=0x200028d8480, visible=<optimized out>)
    at /home/qt/work/qt/qtbase/src/widgets/kernel/qwidget.cpp:8382
#24 0x00005555563b31c9 in MainWindow::MainWindow (this=0x2000024ac00, filenames=...)
    at /local/proyectos/pythonscad/src/gui/MainWindow.cc:946
#25 0x000055555636be88 in gui
    (inputFiles=std::vector of length 1, capacity 1 = {...}, original_path=filesystem::path "/local/proyectos/pythonscad" = {...}, argc=1, argv=0x7fffffffd638)
    at /local/proyectos/pythonscad/src/openscad_gui.cc:248
#26 0x0000555555c0e793 in main (argc=1, argv=0x7fffffffd638)
    at /local/proyectos/pythonscad/src/openscad.cc:1155
(gdb)

gummist, 2025-03-30:
I tried to find the bug, but I'm not familiar enough with the pythonscad sources. 
As an alternative, I compiled the 'jupyter' branch, which is a bit beyond 'master' and bingo, pythonscad works again.
Probably comparing a few source files from both branches will help to pinpoint the problem.

gsohler, 2025-05-18:
i have never seen this mimalloc bug until now. is it still present with the latest sources ?

gummist, 2025-05-19:
I don't know, I'm compiling without mimalloc since back then without any problem. Will check...

gsohler, 2025-12-12:
can we close this issue ?