pythonscad/pythonscad #9: Cannot escape from "Find Handle" view.

KlumpRasmus, 2025-04-18:
**Describe the bug**
Cannot escape "Find Handle" view.

**To Reproduce**
Steps to reproduce the behavior:
1. With a design loaded 
2. Activate "Find Handle" by pressing the circle icon.
3. Try to get back to normal view.

**Expected behavior**
Either a second activation of "Find Handle" button or a toggle behaviour of the button would work (Or pressing escape etc.).

**Code reproducing the issue**
NA

**Screenshots**
NA

**Environment and Version info (please complete the following information):**
OS: Linux 22.04 - Wayland
CPU: Intel Core i5
Open(Python)SCAD: v. 2025.04.05 (homebrew)

**Library & Graphics card information**
Attached

**Additional context**
NA

gsohler, 2025-05-04:
Sorry, I did not notice your issue until now.
it was supposed to work with "Escape"but it was broken.
With latest commit its working again.
will be available on next release

KlumpRasmus, 2025-05-06:
Thanks a lot  :-)

closed 2025-05-18 by gsohler