pythonscad/pythonscad #10: Sefault when using difference()

nomike, 2025-05-20:
**Describe the bug**
When rendering this code, PythonSCAD segfaults:

```python
from openscad import *

obj = difference([cube(20), cube(10)])
show(obj)```

I know the code is not correct, but PythonSCAD should not segfault because of this.

I've also reported this with the OpenSCAD project as <https://github.com/openscad/openscad/issues/5906> but I guess this should be handled upstream.

**Environment and Version info (please complete the following information):**
AppImage

gsohler, 2025-05-20:
Thank you for filing the issue here.
When manifold is upstream to openscad, suppose here is downstream :)

I have tested your example here and i could not find the crash. suppose we best compare versions next.
OTOH i have fixed a nasty double free error yesterday, so my maybe this resolved it. its checked in.
please let me know how you want to proceed
|
[image]

nomike, 2025-05-20:
I could reproduce this with https://www.pythonscad.org/downloads/PythonSCAD-2025-04-02.AppImage.
I'm currently trying to package PythonSCAD for GNU guix but I'm still having some issues there. I probably will open an issue for that as well if I'm not able to figure that out soon.

Once I have it up and running, I can try if I still experience the issue.

nomike, 2025-05-20:
But yes, a double free sounds like a good candidate.

gsohler, 2025-05-21:
Yes, i can replicate the crash in the AppImage, but in the latest version its gone.
Suppose we need an updated AppImage

nomike, 2025-05-21:
I just managed to build pythonscad with commit 39eb62366ef0f49a71d2d6e63441424e75c74690 and the issue is gone.

I guess this should be reported with openscad, as they have to fix it on their end as well.

gsohler, 2025-05-21:
Yes of course,

But dont have any idea, which change has actually resolved the crash. So i cannot create a PR within 5 minutes.
There are quite some PR's already pending, which offer bugfixes and additional features to upstream.
Honestly I dont want to stress these guys with yet another one. In any case they focus on the venv as 1st priority in any case.

BTW: new AppImage is online

nomike, 2025-05-21:
Which commit was the previous appimage based on? I can try to figure out which commit introduced the fix.

gsohler, 2025-05-21:
the commit is 853bfcfbe14b4d954c4385e24a90b8bfc55501e8

its likely the recent bugfix wit the double-free , but i cant be sure

nomike, 2025-05-22:
I'm building that commit right now.
I can confirm, that it does not exit in 54bbcb5, because I have built that already.

If it's happening in 853bfcf, there are 89 commits in between which could be the culprit.
If I do a binary search I'm guaranteed to find the respective commit with 7 or less builds.

But I can probably weed out some of the commits already which should narrow things down significantly.

nomike, 2025-05-22:
https://github.com/pythonscad/pythonscad/commit/853bfcfbe14b4d954c4385e24a90b8bfc55501e8 is having the bug.

nomike, 2025-05-22:
The big was fixed in 618b8179c31ae6e5d6134507b22a9d556eb0aa21.

I will report this to the OpenSCAD project.

gsohler, 2025-05-22:
> The big was fixed in 618b817
>
> I will report this to the OpenSCAD project.

very likely it's the very last diff, but you could user your "binary search
algorithm"  on this file only

nomike, 2025-05-23:
In OpenSCAD this bug no longer exists as well. Seems like the pulled it from pythonscad.
So this ticket is no longer relevant.

closed 2025-05-23 by nomike