pythonscad/pythonscad #11: Wrong shebang in python tests

nomike, 2025-05-20:
As I already have a for of OpenSCAD on my GitHub namespace, GitHub doesn't seem to allow me to create a fork of pyhonscad, so I have to submit my patch like this.

In some python files, the sheband is `#!/usr/bin/env python` instead of `#!/usr/bin/env/python3`.

Here is a[patch to fix this](https://github.com/user-attachments/files/20357515/0001-Use-python3-in-shebang.patch.gz).

Should be fairly straight forward and not in need of any explanation.

nomike, 2025-05-20:
I missed two files. Here is an [updated patch](https://github.com/user-attachments/files/20357599/0001-Use-python3-in-shebang.patch.gz).

gsohler, 2025-05-21:
Its strange.

Creating a fork right here does not work ?

[image: image.png]

nomike, 2025-05-21:
[image]

Nope, doesn't .

gsohler, 2025-05-21:
Its a stupid limitation. In any case, your patch  is applied and pushed. Thank you!

nomike, 2025-05-23:
Issue is solved.