pythonscad/pythonscad #12: 'osimport()' is not relative

nomike, 2025-05-22:
`osimport('foo.svg') searches for `${HOME}/foo.svg` but I would expect it to look in the directory where the pythonSCAD source file is located.

**Environment and Version info (please complete the following information):**
 - OS: GNU Guix

```plaintext
$ openscad-python --info
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 0.0.0-0.b856456 (git b856456fcf26e089a616e5a9b5d685b8a8f2e2c1)
System information: Linux 6.14.0-15-generic #15-Ubuntu SMP PREEMPT_DYNAMIC Sun Apr  6 15:05:05 UTC 2025 x86_64 Ubuntu 25.04 12 CPUs 45.28 GB RAM
User Agent: OpenSCAD/0.0.0-0.b856456 (git b856456fcf26e089a616e5a9b5d685b8a8f2e2c1) (Linux x86_64; Ubuntu 25.04)
Compiler: GCC "11.4.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_83
Eigen version: 3.4.0
CGAL version, kernels: 5.6.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>
Python Version: Python 3.11.11
OpenCSG version: OpenCSG 1.6.0
Clipper2 version: 1.5.2
Manifold version: 3.0.1
Qt version: 5.15.15
QScintilla version: 2.13.4
InputDrivers: 
GLib version: 2.82.1
lodepng version: 20230410
libzip version: 1.9.2
fontconfig version: 2.14.0
freetype version: 2.13.0
harfbuzz version: 8.3.0
cairo version: 1.18.2 (runtime: 1.18.4)
lib3mf version: 2.2.0
Features: roof, input-driver-dbus, skin, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /gnu/store/rnqg1py5d5qb3g3jadimvv0n856aj9d6-pythonscad-0.0.0-0.b856456/bin
Documents Path: /home/nomike/.local/share
User Documents Path: /home/nomike/Documents
Resource Path: /gnu/store/rnqg1py5d5qb3g3jadimvv0n856aj9d6-pythonscad-0.0.0-0.b856456/share/openscad
User Library Path: /home/nomike/.local/share/OpenSCAD/libraries
User Examples Path: /home/nomike/.local/share/OpenSCAD/examples
User Config Path: /home/nomike/.config/OpenSCAD
Backup Path: /home/nomike/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/nomike/.local/share/OpenSCAD/libraries
  /gnu/store/rnqg1py5d5qb3g3jadimvv0n856aj9d6-pythonscad-0.0.0-0.b856456/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /gnu/store/mjg1gq1w6csggjhy5wh6njyzqpic4avq-font-dejavu-2.37/share/fonts
  /home/nomike/.local/share/fonts
  /home/nomike/.local/share/flatpak/exports/share/fonts
  /var/lib/flatpak/exports/share/fonts
  /usr/local/share/fonts
  /usr/share/fonts
  /var/lib/snapd/desktop/fonts
  /gnu/store/rnqg1py5d5qb3g3jadimvv0n856aj9d6-pythonscad-0.0.0-0.b856456/share/fonts
  /gnu/store/fvfzcrgkn4jvqafhrzm8a6kwxzdjpz72-glib-2.82.1/share/fonts
  /home/nomike/.fonts
  /home/nomike/.guix-home/profile/share/fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /home/nomike/Library/Fonts
  /gnu/store/mjg1gq1w6csggjhy5wh6njyzqpic4avq-font-dejavu-2.37/share/fonts/truetype
  /usr/share/fonts/X11
  /usr/share/fonts/cMap
  /usr/share/fonts/cmap
  /usr/share/fonts/fonts-go
  /usr/share/fonts/opentype
  /usr/share/fonts/truetype
  /usr/share/fonts/type1
  /usr/share/fonts/woff
  /home/nomike/.guix-home/profile/share/fonts/opentype
  /home/nomike/.guix-home/profile/share/fonts/truetype
  /usr/share/fonts/X11/Type1
  /usr/share/fonts/X11/encodings
  /usr/share/fonts/X11/misc
  /usr/share/fonts/X11/util
  /usr/share/fonts/cmap/adobe-cns1
  /usr/share/fonts/cmap/adobe-gb1
  /usr/share/fonts/cmap/adobe-japan1
  /usr/share/fonts/cmap/adobe-japan2
  /usr/share/fonts/cmap/adobe-korea1
  /usr/share/fonts/opentype/artemisia
  /usr/share/fonts/opentype/asana-math
  /usr/share/fonts/opentype/cabin
  /usr/share/fonts/opentype/cantarell
  /usr/share/fonts/opentype/comic-neue
  /usr/share/fonts/opentype/complutum
  /usr/share/fonts/opentype/didot
  /usr/share/fonts/opentype/ebgaramond
  /usr/share/fonts/opentype/font-awesome
  /usr/share/fonts/opentype/freefont
  /usr/share/fonts/opentype/inter
  /usr/share/fonts/opentype/ipaexfont-gothic
  /usr/share/fonts/opentype/ipaexfont-mincho
  /usr/share/fonts/opentype/ipafont-gothic
  /usr/share/fonts/opentype/ipafont-mincho
  /usr/share/fonts/opentype/linux-libertine
  /usr/share/fonts/opentype/lobster
  /usr/share/fonts/opentype/lobstertwo
  /usr/share/fonts/opentype/malayalam
  /usr/share/fonts/opentype/neohellenic
  /usr/share/fonts/opentype/noto
  /usr/share/fonts/opentype/olga
  /usr/share/fonts/opentype/porson
  /usr/share/fonts/opentype/roboto
  /usr/share/fonts/opentype/solomos
  /usr/share/fonts/opentype/stix
  /usr/share/fonts/opentype/stix-word
  /usr/share/fonts/opentype/urw-base35
  /usr/share/fonts/truetype/Gargi
  /usr/share/fonts/truetype/Gubbi
  /usr/share/fonts/truetype/Nakula
  /usr/share/fonts/truetype/Navilu
  /usr/share/fonts/truetype/Sahadeva
  /usr/share/fonts/truetype/Sarai
  /usr/share/fonts/truetype/adf
  /usr/share/fonts/truetype/andika
  /usr/share/fonts/truetype/annapurna
  /usr/share/fonts/truetype/arphic
  /usr/share/fonts/truetype/baskerville
  /usr/share/fonts/truetype/charis
  /usr/share/fonts/truetype/clear-sans
  /usr/share/fonts/truetype/comfortaa
  /usr/share/fonts/truetype/croscore
  /usr/share/fonts/truetype/crosextra
  /usr/share/fonts/truetype/dejavu
  /usr/share/fonts/truetype/droid
  /usr/share/fonts/truetype/ebgaramond
  /usr/share/fonts/truetype/font-awesome
  /usr/share/fonts/truetype/fonts-beng-extra
  /usr/share/fonts/truetype/fonts-deva-extra
  /usr/share/fonts/truetype/fonts-gujr-extra
  /usr/share/fonts/truetype/fonts-guru-extra
  /usr/share/fonts/truetype/fonts-kalapi
  /usr/share/fonts/truetype/fonts-orya-extra
  /usr/share/fonts/truetype/fonts-telu-extra
  /usr/share/fonts/truetype/fonts-yrsa-rasa
  /usr/share/fonts/truetype/freefont
  /usr/share/fonts/truetype/gentium
  /usr/share/fonts/truetype/gentium-basic
  /usr/share/fonts/truetype/gentiumplus
  /usr/share/fonts/truetype/gentiumplus-compact
  /usr/share/fonts/truetype/lato
  /usr/share/fonts/truetype/liberation
  /usr/share/fonts/truetype/libreoffice
  /usr/share/fonts/truetype/lohit-assamese
  /usr/share/fonts/truetype/lohit-bengali
  /usr/share/fonts/truetype/lohit-devanagari
  /usr/share/fonts/truetype/lohit-gujarati
  /usr/share/fonts/truetype/lohit-kannada
  /usr/share/fonts/truetype/lohit-malayalam
  /usr/share/fonts/truetype/lohit-oriya
  /usr/share/fonts/truetype/lohit-punjabi
  /usr/share/fonts/truetype/lohit-tamil
  /usr/share/fonts/truetype/lohit-tamil-classical
  /usr/share/fonts/truetype/lohit-telugu
  /usr/share/fonts/truetype/lyx
  /usr/share/fonts/truetype/malayalam
  /usr/share/fonts/truetype/msttcorefonts
  /usr/share/fonts/truetype/noto
  /usr/share/fonts/truetype/open-sans
  /usr/share/fonts/truetype/pagul
  /usr/share/fonts/truetype/paratype
  /usr/share/fonts/truetype/roboto
  /usr/share/fonts/truetype/samyak
  /usr/share/fonts/truetype/samyak-fonts
  /usr/share/fonts/truetype/teluguvijayam
  /usr/share/fonts/truetype/tuffy
  /usr/share/fonts/truetype/ubuntu
  /usr/share/fonts/type1/texlive-fonts-recommended
  /usr/share/fonts/type1/urw-base35
  /usr/share/fonts/woff/ebgaramond
  /usr/share/fonts/X11/encodings/large
  /usr/share/fonts/opentype/roboto/slab
  /usr/share/fonts/truetype/roboto/unhinted
  /usr/share/fonts/truetype/roboto/unhinted/RobotoTTF
```

nomike, 2025-05-23:
I managed reliably reproduce this issue.

Steps to reproduce:
1. Download [PythonSCAD-2025-05-21.AppImage](https://www.pythonscad.org/downloads/PythonSCAD-2025-05-21.AppImage)
2. Download and extract this [simple example](https://github.com/user-attachments/files/20402304/openscad-python-12.zip)
3. Open `star.py` in openscad-python

You will get this:

[image]

It doesn't search for `star.svg` relative to the home directory, but relative to the current working directory of the application.

gsohler, 2025-05-23:
Fixed with c7e7172421ae703bc682274376fcb30349ea771c

its a great addition, thank you!

closed 2025-05-29 by gsohler