pythonscad/pythonscad #14: GL context provider 'glx' not found

nomike, 2025-05-29:
I'm trying to get tests running on the GNU guix package, but almost all of them fail like this:

```plaintext
          Start  842: preview-cgal_triangle-with-duplicate-vertex
 842/2660 Test  #842: preview-cgal_triangle-with-duplicate-vertex ....................***Failed    0.34 sec
triangle-with-duplicate-vertex
test_cmdline_tool: args:['/tmp/guix-build-pythonscad-2025.05.25-0.97c42c9.drv-0/source/tests/test_cmdline_tool.py', '--comparator=image_compare', '-c', '/tmp/guix-build-pythonscad-2025.05.25-0.97c42c9.drv-0/build/tests/venv/bin//python', '-s', 'png', '-e', 'preview', '-t', 'preview-cgal', '-f', 'triangle-with-duplicate-vertex', '/tmp/guix-build-pythonscad-2025.05.25-0.97c42c9.drv-0/build/openscad', '/tmp/guix-build-pythonscad-2025.05.25-0.97c42c9.drv-0/source/tests/data/scad/dxf/triangle-with-duplicate-vertex.scad', '--camera=0,0,100,0,0,0', '--viewall', '--autocenter', '--projection=ortho', '--backend=cgal'] 
test_cmdline_tool: getopt args:['/tmp/guix-build-pythonscad-2025.05.25-0.97c42c9.drv-0/source/tests/test_cmdline_tool.py', '--comparator=image_compare', '-c', '/tmp/guix-build-pythonscad-2025.05.25-0.97c42c9.drv-0/build/tests/venv/bin//python', '-s', 'png', '-e', 'preview', '-t', 'preview-cgal', '-f', 'triangle-with-duplicate-vertex', '/tmp/guix-build-pythonscad-2025.05.25-0.97c42c9.drv-0/build/openscad', '/tmp/guix-build-pythonscad-2025.05.25-0.97c42c9.drv-0/source/tests/data/scad/dxf/triangle-with-duplicate-vertex.scad', '--camera=0,0,100,0,0,0', '--viewall', '--autocenter', '--projection=ortho', '--backend=cgal'] 
run_test() cmdline: /tmp/guix-build-pythonscad-2025.05.25-0.97c42c9.drv-0/build/openscad /tmp/guix-build-pythonscad-2025.05.25-0.97c42c9.drv-0/source/tests/data/scad/dxf/triangle-with-duplicate-vertex.scad --camera=0,0,100,0,0,0 --viewall --autocenter --projection=ortho --backend=cgal -o /tmp/guix-build-pythonscad-2025.05.25-0.97c42c9.drv-0/build/tests/output/preview-cgal/triangle-with-duplicate-vertex-actual.png
using font directory: /tmp/guix-build-pythonscad-2025.05.25-0.97c42c9.drv-0/source/tests/data/ttf
stderr output: Could not initialize localization (application path is '/tmp/guix-build-pythonscad-2025.05.25-0.97c42c9.drv-0/build').
Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
GL context provider 'glx' not found

Error: openscad failed with return code -11
```

I guess there's a dependency missing or something.

Here's my current package definition:

```scheme
(define-public pythonscad
  (let ((commit "97c42c9159777eb41c5d983fda5ac02e035d7742")
        (version "2025.05.25")
        (revision "0"))
    (package
      (inherit openscad)
      (name "pythonscad")
      (version (git-version version revision commit))
      (source
       (origin
         (method git-fetch)
         (uri (git-reference
               (url "https://github.com/pythonscad/pythonscad")
               (commit commit)
               ;; (recursive? #t)
               ))
         (sha256
          (base32 "1ig2qrl6pxqd5bis4gyfiii09k3igg11cgij4lcaqdah0ks94z13"))
         (file-name (git-file-name name version))))
      (arguments
       (list
        #:configure-flags
        #~(list "-DCMAKE_BUILD_TYPE=Release"
                "-DUSE_BUILTIN_OPENCSG=ON"
                "-DMANIFOLD_PYBIND=OFF"
                "-DMANIFOLD_TEST=OFF"
                "-DENABLE_TESTS=ON"
                "-DEXPERIMENTAL=ON"
                "-DSNAPSHOT=ON"
                "-DENABLE_PYTHON=ON"
                (string-append "-DPYTHON_VERSION="
                               #$(version-major+minor (package-version (this-package-input
                                                                        "python"))))
                "-DUSE_BUILTIN_CLIPPER2=OFF"
                (string-append "-DOPENSCAD_VERSION="
                               #$version)
                (string-append "-DOPENSCAD_COMMIT="
                               #$commit)
                "-DENABLE_EGL=ON"
                "-DENABLE_GLX=ON")
        #:phases
        #~(modify-phases %standard-phases
            (add-after 'unpack 'patch-source
              (lambda* (#:key inputs #:allow-other-keys)
                ;; Two of the tests scripts are missing the "#!" prefix on the
                ;; shebang (see
                ;; https://github.com/pythonscad/pythonscad/pull/13).
                (substitute* "tests/test_cmdline_tool.py"
                  (("^/usr/bin/env python3")
                   "#!/usr/bin/env python3"))
                (substitute* "tests/test_pretty_print.py"
                  (("^/usr/bin/env python3")
                   "#!/usr/bin/env python3"))
                ;; The build system fails to find lib3mf; it seems it should
                ;; be using PC_LIBRMF_INCLUDEDIR instead of
                ;; PC_LIB3MF_INCLUDE_DIRS (see
                ;; https://github.com/openscad/openscad/issues/5877).
                (substitute* "cmake/Modules/FindLib3MF.cmake"
                  (("PC_LIB3MF_INCLUDE_DIRS")
                   "PC_LIB3MF_INCLUDEDIR"))
                ;; The build system failes to find EGL; a patch to
                ;; CMakeLists.txt is necessary to find the package
                ;; before adding it to the linked libraries (see
                ;; https://github.com/openscad/openscad/issues/5880).
                (substitute* "CMakeLists.txt"                                    
                  (("target_link_libraries\\(OpenSCAD PRIVATE OpenGL::EGL\\)")
                   "      find_package(ECM REQUIRED NO_MODULE)
      list(APPEND CMAKE_MODULE_PATH ${ECM_MODULE_PATH})
      find_package(EGL REQUIRED)
      target_link_libraries(OpenSCAD PRIVATE EGL::EGL)")
                  ;; Python support is enabled with the -DENABLE_PYTHON=ON
                  ;; build flag.  This requires the Nettle library as a
                  ;; dependency.  The build files however, don´t specify
                  ;; Nettla as mandatory but instead just set ENABLE_PYTHON to
                  ;; off.  This patch set's Nettl to REQUIRED so that the
                  ;; build properly fails if it is missing from the inputs
                  ;; (see https://github.com/openscad/openscad/issues/5897).
                  (("find_package\\(Nettle 3.4\\)")
                   "find_package(Nettle 3.4 REQUIRED)")
                  ;; Use the system sanitizers-cmake module.
                  (("\\$\\{CMAKE_SOURCE_DIR\\}/submodules/sanitizers-cmake/cmake")
                   (string-append (assoc-ref inputs "sanitizers-cmake")
                                  "/share/sanitizers-cmake/cmake")))))
            (add-before 'check 'set-home-directory
              (lambda _
                ;; Tests fail with: Fontconfig error: No writable cache
                ;; directories
                (setenv "HOME" "/tmp"))))))
      (inputs (modify-inputs (package-inputs openscad)
                (append libjpeg-turbo curl python nettle)))
      (home-page "https://pythonscad.org/")
      (description
       "PythonSCAD provides a powerful framework for defining 3D
models using the Python programming language.  It extends the
capabilities of OpenSCAD, a code-driven 3D CAD modeller, by overcoming
OpenSCAD's native language limitations such as variable immutability,
limited iterations, and lack of file I/O.  This integration enables
more complex parametric designs, leverages Python's extensive library
ecosystem for data manipulation and calculations, and promotes an
object-oriented approach to 3D modeling."))))
```

Anyone has any idea about what could be wrong here?

Thanks!

nomike

gsohler, 2025-05-29:
I think the error comes from

https://github.com/pythonscad/pythonscad/blob/master/src/glview/OffscreenContextFactory.cc#L106
I think you should make sure, that ENABLE_GLX is set in your build.

not nice anyway that this causes a segfault

nomike, 2025-05-29:
It is enabled already. That's what's odd.

If it's enabled and it builds alright, it should find it. 

If it doesn't find it, the configure stage should fail already. 

I will have a look at this later today.

gsohler, 2025-05-29:
are you really sure that it *is* and *is finally* enabled and not turned off by CMakeLists for *any* reason ?
I turned it off explicitely and i exactly face your error.

nomike, 2025-05-29:
Good point. Those CMake scripts have an odd tendency to not fail the build if crucial libraries are missing for a feature, but too disabling the feature flag which was EXPLICITLY ENABLED before. ¯\_(ツ)_/¯

I will have a look when I'm back home and if that's the case I will send a patch to OpenSCAD...

nomike, 2025-05-30:
I've filed a few tickets and PRs with OpenSCAD:

- https://github.com/CGAL/cgal/issues/8914
- https://github.com/openscad/openscad/pull/5933
- https://github.com/openscad/openscad/pull/5934
- https://github.com/openscad/openscad/pull/5935

I've also filed a ticket with CGAL, which also has an issue with CMake:

- https://github.com/CGAL/cgal/issues/8914

But that was nothing serious yet.

I'm getting to the sauce now I guess.

nomike, 2025-06-24:
@gsohler can you please try to build PythonSCAD with the following patch applied?

And if you get the same error message, we need to figure out how to fix the build with this patch applied. Because I pretty much think that this would fix the build on guix.

Another option would perhaps be to get GLEW running, but that seems to be more complicated.

I'm way to unfamiliar with OpenSCAD/PythonSCAD and GLEW to solve this I'm afraid...