pythonscad/pythonscad #19: naming conflict with OpenSCAD

nomike, 2025-06-06:
Both OpenSCAD and pythonscad produce executables named `openscad` and `openscad-python`.

This makes it very hard (guix) or even impossible (Ubuntu, Debian, Fedora,....probably almost all other Linux distributions) to install both at the same time.

I'm not sure why openscad has a openscad-python executable, I will probably file an issue with them as well to address this.

However, as this project here is called pythonscad, I suggest also naming the binary like that to avoid that conflict.

nomike, 2025-06-06:
As discussed in the OpenSCAD ticket, both the openscad and the openscad-python executable are needed (the latter for venv setup).

So I think the only way forward would be to rename the excutables of this project to `pythonscad` (and pythonscad-python` if the same requirement for venv setup applies here).

I would still be able to package pythonscad for gnu guix I guess, though there might be discussions with the OS maintainers about that, as it's far from ideal that two packages can not be installed together at the same time.
But I also assume, that it will be hard to have a package accepted by the debian and Fedora projects. 

I think a good way forward would be to parametrize this in `/CMakeLists.txt` with a "-D" flag. This way we might apply the same commit on both OpenSCAD and pythonscad and thus avoid unnecessary merge conflicts in the future.

If this is something you want, I can have a look into that, come up with a patch and discuss it with the OpenSCAD folks as well.

gsohler, 2025-06-06:
Yes, a CMakeList parameter is a great 1st step, but i doubt, it will solve all issues
(openscad might be hardcoded in many other places)
I know that you cant fork pythonscad, but it would be so useful. I could checkout your branch in your github to test 1st
(maybe have another guthub account ?)
BTW: did you read your matrix ?

nomike, 2025-07-20:
This has since been fixed by changing the executable name from "openscad" to "pythonscad".

closed 2025-07-20 by nomike