pythonscad/pythonscad #20: Add `bounding_box()` and `position()` methods to `PyOpenSCAD` class

nomike, 2025-06-08:
It would be great if objects had convenience methods for getting the bounding box and the position/coordinates.

I've implemented it like this for now, but I think it would be nicer to have this built in.

```python
def get_bounding_box(object):
    min_x = min([point[0] for point in object.mesh()[0]])
    max_x = max([point[0] for point in object.mesh()[0]])
    min_y = min([point[1] for point in object.mesh()[0]])
    max_y = max([point[1] for point in object.mesh()[0]])
    min_z = min([point[2] for point in object.mesh()[0]])
    max_z = max([point[2] for point in object.mesh()[0]])
    return (
        max_x - min_x,
        max_y - min_y,
        max_z - min_z
    )

def get_object_coordinates(object):
    min_x = min([point[0] for point in object.mesh()[0]])
    min_y = min([point[1] for point in object.mesh()[0]])
    min_z = min([point[2] for point in object.mesh()[0]])
    return (
        min_x,
        min_y,
        min_z
    )
```

nomike, 2025-06-08:
Another neat function I've just implemented:

```python
def position_at_origin(object):
    coordinates = get_object_coordinates(object)
    return object.translate([-coordinates[0], -coordinates[1], -coordinates[2]])
```

nomike, 2025-06-08:
I'm currently designing a universal parametric phone cover and I do come up with new ideas all the time.

If someone can show me how to implement the `get_position()` function (also for object oriented mode) as an example (a patch would be great), I could have a look at other functions too and create a merge request for them.

gsohler, 2025-06-08:
Getting the bbox of a shape is indeed used very often and i think an integrated solution is helpful even though it can
be implemented with mesh.
I will create a bbox() function which will return a tuple of min and max coordinates.
Your sample code is very ineffective because mesh() is called 6 times. mesh is an expensive function.
If you want to center an imported stl instead, import from openscad has a center function(i think osimport should also have)

you could do like position_at_origin function like

minpos, maxpos = obj.bbox()
obj_at_origin = obj - minpos # this line works yet already

will let you knowwhen bbox is in.

gsohler, 2025-06-08:
[image]
This was an easy one.  its already pushes.
A possible extension could be adding 2D objects in addition