pythonscad/pythonscad #24: Issue #9 not fixed

KlumpRasmus, 2025-06-13:
**Describe the bug**
Still cannot escape from "Find Handle" view. See issue #9 

**Environment and Version info (please complete the following information):**
- OS: Linux Ubuntu 22.04
- System: 13th Gen Intel(R) Core(TM) i5-1345U
- OpenSCAD Version: **2025.06.11** (local build)

gsohler, 2025-06-13:
sorry,
I did close #9 because it was working fine for in my place for a long time.
I will investigate, to make it work in your place, too

* I pressed ESC key and the Handle mode was completed, did you try that ?

OR:

* How did you expect to make the command complete ? What is intuitive to
the user ?

KlumpRasmus, 2025-06-14:
Could it have something to do with the fact that I’m using Linux on a laptop where the ESC key also functions as an FnLock key?"

Would it be possible to make the "Find Handles" button a toggle button so it enables/disables the "show Handles" mode?

gsohler, 2025-06-14:
Yes, Indeed!
Thats a wonderful idea. I have already implemented, watch here:

https://www.youtube.com/watch?v=8mnvmIWFfhA

@t-paul maybe this Idea is also applicable to OpenSCAD ?

gsohler, 2025-06-17:
The new Windows installer is online

gsohler, 2025-07-06:
is it fixed for you ?

KlumpRasmus, 2025-07-08:
Weirdly it does still not seem to work. I did a build today on Linux using these commands:

```
git clone https://github.com/pythonscad/pythonscad.git
cd pythonscad
git submodule update --init --recursive
sudo ./scripts/uni-get-dependencies.sh
mkdir build
cd build
cmake -DEXPERIMENTAL=1 -DENABLE_PYTHON=1 -DPYTHON_VERSION=3.12 ..
make -j8
sudo make install
```

When I press the "Find handles" button handles appear, but when I press it again, nothing happens, and ESC key does nothing either.

KlumpRasmus, 2025-07-08:
Git gives me: `294fbf6e2 (HEAD -> master, origin/master, origin/HEAD) enable improtong loval libs again`

I just tried building again, and starting with ./pythonscad, and now it works  - both ESC and the toggle button :-)

It seems I build one thing and install another to the system so the problem is on my end.

Thank you for the help.

gsohler, 2025-07-09:
great!
you spotted my comment which i punched in in 1 sec  without checking

Exactly, I believe that you compiled the most recent one locally, but
always checked with the global installed one.

closed 2025-07-14 by gsohler