pythonscad/pythonscad #25: How to download it?

ZUIcat, 2025-06-15:
How can I download it? Apologies for asking such a basic question, but there's nothing in the releases, and the link in the README (https://www.pythonscad.org/donloads.html) appears to be invalid.
And, do I need to install Python separately, or does it come with an embedded Python interpreter? Which version of the Python is supported?

gsohler, 2025-06-15:
Hi ZUIcat

This was a stupid error in the README.md file. It's now corrected.
the correct link is

https://pythonscad.org/download.php

PythonSCAD comes with an embedded python interpreter, but for stdlib it
still needs an external python interpreter.
It's clearly documented in the webpage for your platform.
We try to unlock this dependency ASAP - already working on it.

Hope this helps

ZUIcat, 2025-06-16:
Thank you for your reply. I downloaded *PythonSCAD_nolibfive-2025.06.05-x86-64-Installer.exe* but didn't install Python 3.11 (since I already have 3.10 installed), and it failed to initialize Python. I briefly looked at some code, like `initPython`. You did embed Python such as `#include <Python.h>`, but without including stdlib, causing the initialization to fail. To quickly test my idea, I copied Python 11's Lib folder to the program's root directory, and it worked (just for verification—I know this isn't the proper approach). May I ask if there's a specific consideration for this?

> Hi ZUIcat
>
> This was a stupid error in the README.md file. It's now corrected.
> the correct link is
>
> https://pythonscad.org/download.php
>
> PythonSCAD comes with an embedded python interpreter, but for [...]

gsohler, 2025-06-16:
Hi ZUIcat,

Thank you for your answer.
The Choice to use exactly  Python 3.11.5-2 is another one

PythonSCAD(OpenSCAD) used MXE to create windows executables.
As Python package is not available in MXE, I managed to  link
python from mingw32 into the application and it worked with luck.
There are only few python versions available with mingw32 and  3.11.5-2  is
one of them.

Can you please elaborate greatly on :

You did embed Python such as #include <Python.h>, but without including
stdlib, causing the initialization to fail.
I think this will help me greatly to improve pythonscad

Thank you

ZUIcat, 2025-06-16:
By the way, unlike SolidPython, PythonSCAD's Python code doesn't work well when opened in external editors because they don't use PythonSCAD's Python environment and can't find custom libraries.  

I think it might be better if PythonSCAD directly includes a complete embedded Python (with python.exe), and places custom libraries like openscad in standard locations (e.g., Lib\site-packages). This way, external editors (like VSCode) can directly set this embedded Python as the interpreter for the code.  

Just a passing thought—haven't deeply considered or tested it.

gsohler, 2025-06-16:
Previously I was actually using the embeddedable  python package.(was another zip file) where i unpackacked the archive.
But I realized, that it did not run 100% stable( help command crashed, and also opencascade import crashed).
This is why i ended up with this version.
I also tried to ship with the contents of stdlib dir in the past. but contents were so big, so i did not follow that path.

If shipping with python311.zip instead does not consume tons of space. I think i could follow that way. it appears lovely to me!

BTW: its indeed possible to use external libraries with pythonscad(with some care about versions).
Having an embedded python instead of a passed-through python interpreter has one major advantage:

Python can directly access and alter the internals of Python(OpenSCAD) and this retrieve back vertex information.(at least its way easier). 

I will try the python311.zip before this ticket can close

gsohler, 2025-06-17:
> it works like a charm. its implemented in commit eb42339
>
> Thank you ZUICat!
>
> I’m glad this information was helpful to you.
>
> Based on the above discussion, I think a few points may need further
> clarification. You might already know some of them, but others could be
> new, which should benefit the project.
>
> The following content is based on my environment: Windows 10 + Python
> 3.10.11.
>
> *Difference Between Normally Installed Python and Embedded Python*
>
> A normal Python installation includes a complete environment: uncompiled
> standard libraries (in the Lib folder), C++ modules for standard
> libraries (in DLLs), Pip (in Scripts), documentation, and other
> components.
>
> Embedded Python is a minimal environment:
>
>    - Core libraries are precompiled into .pyc files and compressed into
>    python310.zip.
>    - C++ modules for standard libraries are scattered in the root
> [... 131 lines cut ...]
> ------------------------------
>
> P.S.: What does the "Create Python Virtual Environment" feature in the
> PythonSCAD do? Does it run python -m venv? I haven’t checked the code
> yet, but it crashes immediately on my computer.

Hi  Zuicat, yes indeed very helpful

Your python knowledge is huge, it appears to me that you must either be a
python core programmer, or python is part of your daily work.

No, the information given above is mostly new to me.
Yes I accept to add further dll and pyc files to the installation, if it
makes sense.
For sqlite3 I doubt that there is any common use with geometric design.

You share a lot of valuable internal  python information and all of
them appear to target
better PythonSCAD module usage.
I am just nor sure if I can already benefit from it, but I will try that.

I'd love to add module management into PythonSCAD, that means

* add pip to embedded python
* add a user function to pip install selected python modules to the
embedded python only(not the system one)

Apart from that it would be great when you could support me by implementing
this - this could be milestones, microsteps, code snippets, pull-requests,
whatever.

However I am not sure if the venv strategy is the correct one. I think the
python3.10.pth approach and maybe get-pip is the better one.

The "Create Virtual Environment|" stuff is created by t-paul. But I
don't use it because it does not work for me nor do I understand it.

In any case, I think my next goal is to find such pth files, I did not
notice it yet, and pythonscad was working even without it  and without any
python installed.

PS Thank you again for your valuable information.

gsohler, 2025-06-17:
BTW I tried quite some time today using
PyWideStringList_Append, but whenever I tried,  Py_InitializeFromConfig would result in a segfault later.

gsohler, 2025-06-18:
I added some module install functionality mainly in

https://github.com/pythonscad/pythonscad/blob/own-pip/src/python/pymod.cc

its somewhat working (for numpy)  but still failing for many other modules

ebo, 2026-03-14:
I think we can probably close this issue.