pythonscad/pythonscad #27: Crashing when trying to access information about object in an array

alacritous13, 2025-06-25:
When trying to access object being stored in an array, after the array has been shown, it causes OpenSCAD to crash. 

```
from openscad import *

c=cube(5)

c1=c;
c2=c.translate([6,0,0]);

ted=[c1,c2];

show(ted); #this causes the crash, commenting it out and it will work fine.
fred=ted[1];

fred.show()
print(fred.origin)
```

This code is just one example of how I managed to crash it, but it all seams to be a variation of:

1. Having an array of objects.
2. Showing the array
3. Trying to access some sub portion of the array 

Best I can guess, the array is being converted or corrupted during the show process.

I'm running Windows version 2025.06.04

gsohler, 2025-06-25:
Thank you, I can confirm the crash
Limited it to

c=cube(5)
ted=[c];
show(ted); #this causes the crash, commenting it out and it will work fine.
ted[0].show()

gsohler, 2025-06-25:
No, arrays are not required.

It literally crashes, when outputting the same shape twice.
from openscad import *

c=cube(5)
d=c
show(c); #this causes the crash, commenting it out and it will work fine.
show(d)

these crashes usually appear when memory is accessed which was freed before but I did not yet find the perfect recipe to combine/marry  python's reference counter with the std::shared_ptr usage counter

Any help appreciated

alacritous13, 2025-06-26:
The example you gave works once then causes a crash.

```
c=cube(5)
print(c[1])
```
This snippet of code crashes the program reliably and repeatably, and I think is what's causing my crash. It only works with numbers, sub-variable names will cause proper error messages.

```
g=cube(5);
g=g.translate([5,5,0])
q=g.translate([5,5,0])
ted=[q,g]
show(ted)
print(ted)
```
If I first print `g` and then print `ted` I can get a value without it crashing (doesn't look to work in the inbuilt editor). The value if it isn't shown is `[OpenSCAD (4), OpenSCAD (7)]`. When it is shown (and it doesn't crash) the value is `[[...]]`.

I've additionally updated to version 2025.06.17

gsohler, 2025-06-26:
Issue found!

https://github.com/pythonscad/pythonscad/blob/master/src/python/pyfunctions.cc#L1923

When show returns back the argument it must increase its reference count because the argument is just borrowed.
When failing to do so, the function receiving the return value will free it and the argument  cannot be reused.

alacritous13, 2025-06-28:
I've loaded the new installer. Problem looks to have been fixed. Thank you.

closed 2025-06-28 by alacritous13