pythonscad/pythonscad #28: Failed test 43 - echo_function-literal-tests

nomike, 2025-06-25:
**Describe the bug**
I'm packaging PythonSCAD for GNU guix (#14) and this test fails.

**Output**
[echo_function-literal-tests.log.txt](https://github.com/user-attachments/files/20913518/echo_function-literal-tests.log.txt)

nomike, 2025-06-26:
Build log:

```plaintext
running 'cmake' with arguments ("../source" "-DCMAKE_BUILD_TYPE=RelWithDebInfo" "-DCMAKE_INSTALL_PREFIX=/gnu/store/2dvqbsl91s87dgqanywm15jb2ib6y14y-pythonscad-2025.06.25-0.0334448" "-DCMAKE_INSTALL_LIBDIR=lib" "-DCMAKE_INSTALL_RPATH_USE_LINK_PATH=TRUE" "-DCMAKE_INSTALL_RPATH=/gnu/store/2dvqbsl91s87dgqanywm15jb2ib6y14y-pythonscad-2025.06.25-0.0334448/lib" "-DCMAKE_VERBOSE_MAKEFILE=ON" "-DCMAKE_BUILD_TYPE=Release" "-DUSE_BUILTIN_CLIPPER2=OFF" "-DUSE_BUILTIN_MANIFOLD=OFF" "-DUSE_BUILTIN_OPENCSG=OFF" "-DMANIFOLD_PYBIND=OFF" "-DMANIFOLD_TEST=OFF" "-DEXPERIMENTAL=ON" "-DENABLE_PYTHON=ON" "-DOPENSCAD_VERSION=2025.06.25-0.0334448" "-DOPENSCAD_COMMIT=03344486e7417fc585c3c16353c776d36b6fb020" "-DENABLE_EGL=ON" "-DENABLE_GLX=ON")
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /gnu/store/a3lsdsalcmg5wnk67869af7wljprkbam-pkg-config-0.29.2/bin/pkg-config (found version "0.29.2") 
-- Eigen: 3.4.0
-- Found Boost: /gnu/store/di4520fl1q7lkxahblsvpd2079zhx0q4-boost-1.83.0/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.61") found components: system regex program_options 
-- Boost: 1.83.0
-- Harfbuzz: 8.3.0
-- Fontconfig: 2.14.0
-- Checking for one of the modules 'glib-2.0'
-- Glib: 2.82.1
-- OPENCSG_DIR: 
-- OpenCSG include found in /gnu/store/29965zfc11j10njskf9lmx8196b05lys-opencsg-1.8.1/include
-- OpenCSG library found in /gnu/store/29965zfc11j10njskf9lmx8196b05lys-opencsg-1.8.1/lib/libopencsg.so
-- OpenCSG: 1.8.1
-- USE_GLAD/USE_GLEW not specified: Defaulting to GLAD
-- OpenGL: /gnu/store/mc0l896lmq74ak7gy55s1b4z68h19440-mesa-25.1.3/lib/libGL.so;/gnu/store/29kjd06xf7140manr0piyzvhx3hj4aq5-glu-9.0.2/lib/libGLU.so
-- Using header-only CGAL
-- Targeting Unix Makefiles
-- Using /gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/bin/c++ compiler.
-- Found GMP: /gnu/store/kka705681m1hq98b9jz98vxk9s5qd4ld-gmp-6.3.0/lib/libgmp.so
-- Found MPFR: /gnu/store/pz4xkjl28w0pqf8nhgv68xix17ap6dsx-mpfr-4.2.1/lib/libmpfr.so
-- Found Boost: /gnu/store/di4520fl1q7lkxahblsvpd2079zhx0q4-boost-1.83.0/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.66")  
-- Boost include dirs: /gnu/store/di4520fl1q7lkxahblsvpd2079zhx0q4-boost-1.83.0/include
-- Boost libraries:    
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Using gcc version 4 or later. Adding -frounding-math
-- CGAL: 5.6.1
-- CGAL: Using target CGAL::CGAL CGAL::CGAL_Core
-- libzip: 1.9.2
-- Freetype: 2.13.0
-- LibXml2: 2.9.14
-- Found Nettle: /gnu/store/al613p11xv5w1xmnqn7ykw0x6d4b0539-nettle-3.9.1/lib/libnettle.so (found suitable version "3.9.1", minimum required is "3.4")
-- Found Python: /gnu/store/s6hivi3vwc1cyyw853y9c8xwahlqym5j-python-3.11.11/include/python3.11 (found version "3.11.11") found components: Development Development.Module Development.Embed
-- Python enabled
-- HidAPI: 0.14.0
-- SpNav: found
-- CAIRO_INCLUDEDIR: /gnu/store/gh9cgprk2x0zq7xw44dal0v8s3wixpj3-cairo-1.18.2/include
-- CAIRO_INCLUDE_DIRS: /gnu/store/7ah6i5829f5ha6bdfzcj1gf21115xyyd-freetype-2.13.0/include/freetype2;/gnu/store/zv420d7jscbcwkzndxb9n9dma4xhmf43-pixman-0.42.2/include/pixman-1;/gnu/store/gh9cgprk2x0zq7xw44dal0v8s3wixpj3-cairo-1.18.2/include/cairo;/gnu/store/qj9jhgqn2b9g7yxdhyw6ki1vz241apkd-libpng-1.6.39/include/libpng16
-- CAIRO_LIBRARIES: cairo
-- Cairo: 1.18.2
-- Flex: 2.6.4
-- Found CURL: /gnu/store/cjvaagi44q7r0nhx49r91kmi96xs6qfh-curl-8.6.0/lib/libcurl.so (found version "8.6.0")
-- curl: found in /gnu/store/cjvaagi44q7r0nhx49r91kmi96xs6qfh-curl-8.6.0/include
-- Bison: 3.8.2
-- Found Lib3MF: 3mf;zip;z (found version "2.2.0") 
-- lib3mf: 2.2.0
-- Qt5: 5.15.15
-- QScintilla: 2.13.4
-- DBus input driver enabled
-- Qt5Gamepad input driver enabled
CMake Warning (dev) at /gnu/store/jkrx0i1b5vz3bakc49zy7kcfl7v1cgag-extra-cmake-modules-6.5.0/share/ECM/modules/ECMFindModuleHelpers.cmake:111 (message):
  Your project should require at least CMake 3.16.0 to use FindEGL.cmake
Call Stack (most recent call first):
  /gnu/store/jkrx0i1b5vz3bakc49zy7kcfl7v1cgag-extra-cmake-modules-6.5.0/share/ECM/find-modules/FindEGL.cmake:45 (ecm_find_package_version_check)
[... 47 lines cut ...]
-- Headless build:        OFF
-- NULLGL build:          OFF
-- OpenGL wrangler:       GLAD
-- Profiling/coverage:    OFF
-- Use mimalloc:          ON
-- Use built-in OpenCSG:  OFF
-- CSG node id-prefix:    OFF
-- Offline documentation: OFF
--  
-- GL Offscreen Context:  Unix EGL
--  
-- Input Drivers
-- - HID API Driver (enabled)
-- - Space Navigator Library Driver (enabled)
-- - Joystick Driver (enabled)
-- - DBus Remote Driver (enabled)
-- - Python support ( enabled )
--  
[... cut ...] DGET;ENABLE_DBUS;ENABLE_QGAMEPAD;OPENSCAD_VERSION=2025.06.25-0.0334448;OPENSCAD_SHORTVERSION=2025.06.25;OPENSCAD_YEAR=2025;OPENSCAD_MONTH=6;OPENSCAD_DAY=25;OPENSCAD_DAY=25;OPENSCAD_COMMIT=03344486e7417fc585c3c16353c776d36b6fb020;STACKSIZE=8388608;OPENSCAD_OS="Unix";ENABLE_EGL;USE_MANIFOLD_TRIANGULATOR;ENABLE_JOYSTICK;ENABLE_MANIFOLD
-- LINK OPTIONS = -v
--  
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/guix-build-pythonscad-2025.06.25-0.0334448.drv-0/build
```

gsohler, 2025-06-26:
if I disable ENABLE_PHYTON, I will exactly get our error. let me find the reason for that ...

https://github.com/pythonscad/pythonscad/blob/master/src/core/Expression.cc#L595
should be excecuted to not show the line

nomike, 2025-06-27:
The most recent updates fixed a number of failed tests.
However, this one still fails for me.

Output from CMake:

```plaintext
source directory: "/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source" (relative from build: "../source")
build directory: "/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/build"
running 'cmake' with arguments ("../source" "-DCMAKE_BUILD_TYPE=RelWithDebInfo" "-DCMAKE_INSTALL_PREFIX=/gnu/store/f5h4dxbbi7pgw1b9igpvyphqs03z62yb-pythonscad-2025.06.26-0.24bcd09" "-DCMAKE_INSTALL_LIBDIR=lib" "-DCMAKE_INSTALL_RPATH_USE_LINK_PATH=TRUE" "-DCMAKE_INSTALL_RPATH=/gnu/store/f5h4dxbbi7pgw1b9igpvyphqs03z62yb-pythonscad-2025.06.26-0.24bcd09/lib" "-DCMAKE_VERBOSE_MAKEFILE=ON" "-DCMAKE_BUILD_TYPE=Release" "-DUSE_BUILTIN_CLIPPER2=OFF" "-DUSE_BUILTIN_MANIFOLD=OFF" "-DUSE_BUILTIN_OPENCSG=OFF" "-DMANIFOLD_PYBIND=OFF" "-DMANIFOLD_TEST=OFF" "-DEXPERIMENTAL=ON" "-DENABLE_PYTHON=ON" "-DOPENSCAD_VERSION=2025.06.26-0.24bcd09" "-DOPENSCAD_COMMIT=24bcd0942a43c8e578074f8be4a1383fa60c8276" "-DPYTHON_VERSION=3.11" "-DENABLE_EGL=ON" "-DENABLE_GLX=ON")
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /gnu/store/a3lsdsalcmg5wnk67869af7wljprkbam-pkg-config-0.29.2/bin/pkg-config (found version "0.29.2") 
-- Eigen: 3.4.0
-- Found Boost: /gnu/store/di4520fl1q7lkxahblsvpd2079zhx0q4-boost-1.83.0/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.61") found components: system regex program_options 
-- Boost: 1.83.0
-- Harfbuzz: 8.3.0
-- Fontconfig: 2.14.0
-- Checking for one of the modules 'glib-2.0'
-- Glib: 2.82.1
-- OPENCSG_DIR: 
-- OpenCSG include found in /gnu/store/29965zfc11j10njskf9lmx8196b05lys-opencsg-1.8.1/include
-- OpenCSG library found in /gnu/store/29965zfc11j10njskf9lmx8196b05lys-opencsg-1.8.1/lib/libopencsg.so
-- OpenCSG: 1.8.1
-- USE_GLAD/USE_GLEW not specified: Defaulting to GLAD
-- OpenGL: /gnu/store/mc0l896lmq74ak7gy55s1b4z68h19440-mesa-25.1.3/lib/libGL.so;/gnu/store/29kjd06xf7140manr0piyzvhx3hj4aq5-glu-9.0.2/lib/libGLU.so
-- Using header-only CGAL
-- Targeting Unix Makefiles
-- Using /gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/bin/c++ compiler.
-- Found GMP: /gnu/store/kka705681m1hq98b9jz98vxk9s5qd4ld-gmp-6.3.0/lib/libgmp.so
-- Found MPFR: /gnu/store/pz4xkjl28w0pqf8nhgv68xix17ap6dsx-mpfr-4.2.1/lib/libmpfr.so
-- Found Boost: /gnu/store/di4520fl1q7lkxahblsvpd2079zhx0q4-boost-1.83.0/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.66")  
-- Boost include dirs: /gnu/store/di4520fl1q7lkxahblsvpd2079zhx0q4-boost-1.83.0/include
-- Boost libraries:    
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Using gcc version 4 or later. Adding -frounding-math
-- CGAL: 5.6.1
-- CGAL: Using target CGAL::CGAL CGAL::CGAL_Core
-- libzip: 1.9.2
-- Freetype: 2.13.0
-- LibXml2: 2.9.14
-- Found Nettle: /gnu/store/al613p11xv5w1xmnqn7ykw0x6d4b0539-nettle-3.9.1/lib/libnettle.so (found suitable version "3.9.1", minimum required is "3.4")
-- Found Python: /gnu/store/s6hivi3vwc1cyyw853y9c8xwahlqym5j-python-3.11.11/include/python3.11 (found suitable exact version "3.11.11") found components: Development Development.Module Development.Embed
-- Python enabled
-- HidAPI: 0.14.0
-- SpNav: found
-- CAIRO_INCLUDEDIR: /gnu/store/gh9cgprk2x0zq7xw44dal0v8s3wixpj3-cairo-1.18.2/include
-- CAIRO_INCLUDE_DIRS: /gnu/store/7ah6i5829f5ha6bdfzcj1gf21115xyyd-freetype-2.13.0/include/freetype2;/gnu/store/zv420d7jscbcwkzndxb9n9dma4xhmf43-pixman-0.42.2/include/pixman-1;/gnu/store/gh9cgprk2x0zq7xw44dal0v8s3wixpj3-cairo-1.18.2/include/cairo;/gnu/store/qj9jhgqn2b9g7yxdhyw6ki1vz241apkd-libpng-1.6.39/include/libpng16
-- CAIRO_LIBRARIES: cairo
-- Cairo: 1.18.2
-- Flex: 2.6.4
-- Found CURL: /gnu/store/cjvaagi44q7r0nhx49r91kmi96xs6qfh-curl-8.6.0/lib/libcurl.so (found version "8.6.0")
-- curl: found in /gnu/store/cjvaagi44q7r0nhx49r91kmi96xs6qfh-curl-8.6.0/include
-- Bison: 3.8.2
-- Found Lib3MF: 3mf;zip;z (found version "2.2.0") 
-- lib3mf: 2.2.0
-- Qt5: 5.15.15
-- QScintilla: 2.13.4
-- DBus input driver enabled
[... 105 lines cut ...]
+WARNING: Ignoring unknown function '(true)' in file function-literal-tests.scad, line 6
 ECHO: undef
 WARNING: Can't call function on number in file function-literal-tests.scad, line 7
+WARNING: Ignoring unknown function '(10.5)' in file function-literal-tests.scad, line 7
 ECHO: undef
 WARNING: Can't call function on string in file function-literal-tests.scad, line 8
+WARNING: Ignoring unknown function '("string")' in file function-literal-tests.scad, line 8
 ECHO: undef
 WARNING: Can't call function on vector in file function-literal-tests.scad, line 9
+WARNING: Ignoring unknown function '([])' in file function-literal-tests.scad, line 9
 ECHO: undef
 WARNING: Can't call function on range in file function-literal-tests.scad, line 10
+WARNING: Ignoring unknown function '([1 : 4])' in file function-literal-tests.scad, line 10
 ECHO: undef
 ECHO: is_function1 = false
 ECHO: is_function2 = true

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   1.16 sec

The following tests FAILED:
         43 - echo_function-literal-tests (Failed)
Errors while running CTest
```

gsohler, 2025-06-27:
As noted yesterday, here its clean

Running '/home/gsohler/git/pythonscad/build/openscad --info' to generate sysinfo.txt
QObject::startTimer: Timers can only be used with threads started with QThread
Post test:/usr/bin/python3.12 "/home/gsohler/git/pythonscad/tests/test_pretty_print.py" --builddir=/home/gsohler/git/pythonscad/build 
Test project /home/gsohler/git/pythonscad/build
    Start 43: echo_function-literal-tests
1/1 Test #43: echo_function-literal-tests ......   Passed    8.08 sec

100% tests passed, 0 tests failed out of 1

Total Test time (real) =   8.24 sec
report saved:
 Testing/Temporary/__svga3d;-build:-release;-_tabr_report.html

=====================================================

When I change the code in src/core/Expression.cc (595) to disable the code like that and compile

#ifdef ENABLE_PYTHON_DISABLED
      if(f == boost::none)
      {
        int error;	      
        Value v = python_functionfunc(call,context, error);
        if(!error) return v;
      }
#endif    

=========================================================
Then the test fails exactly like in your place

@ -2,14 +2,19 @@

 WARNING: Ignoring unknown function '(undef)' in file function-literal-tests.scad, line 5
 ECHO: undef
 WARNING: Can't call function on bool in file function-literal-tests.scad, line 6
+WARNING: Ignoring unknown function '(true)' in file function-literal-tests.scad, line 6
 ECHO: undef
 WARNING: Can't call function on number in file function-literal-tests.scad, line 7
+WARNING: Ignoring unknown function '(10.5)' in file function-literal-tests.scad, line 7
 ECHO: undef
 WARNING: Can't call function on string in file function-literal-tests.scad, line 8
+WARNING: Ignoring unknown function '("string")' in file function-literal-tests.scad, line 8
 ECHO: undef
 WARNING: Can't call function on vector in file function-literal-tests.scad, line 9
+WARNING: Ignoring unknown function '([])' in file function-literal-tests.scad, line 9
 ECHO: undef
 WARNING: Can't call function on range in file function-literal-tests.scad, line 10
+WARNING: Ignoring unknown function '([1 : 4])' in file function-literal-tests.scad, line 10
 ECHO: undef
 ECHO: is_function1 = false
 ECHO: is_function2 = true

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   8.24 sec

The following tests FAILED:
	 43 - echo_function-literal-tests (Failed)
report saved:
 Testing/Temporary/__svga3d;-build:-release;-_tabr_report.html
Errors while running CTest

=================================================================
Now I am concluding that ENABLE_PYTHON is NOT set for src/core/Expression.cc for unknown reason

you could add an #error statement into the ifdef and your compilatin must fail then(but I think it does not)

nomike, 2025-07-02:
I've added a patch to the package definition which adds this line

```c
#error "This should crash!"
```
[after this line](https://github.com/pythonscad/pythonscad/blob/97b3a0e7de604d53fcf9a7af49aedd0164ba3f79/src/core/Expression.cc#L595).

And indeed, gcc shows this error:

```plaintext
/tmp/guix-build-pythonscad-2025.07.01-0.d5a2b6a.drv-0/source/src/core/Expression.cc:596:2: error: #error "This should crash"
  596 | #error "This should crash"
      |  ^~~~~
```

gsohler, 2025-07-03:
I think that I have solved the issue. it happened due to different behaviour for unitialized variables