pythonscad/pythonscad #29: Failing tests 112 - echo_errors-warnings and 113 - echo_errors-warnings-included

nomike, 2025-06-25:
**Describe the bug**
When building PythonSCAD on GNU guix the test case fails.

**Output**

[echo_errors-warnings.log.txt](https://github.com/user-attachments/files/20913667/echo_errors-warnings.log.txt)

[echo_errors-warnings-included.log.txt](https://github.com/user-attachments/files/20913680/echo_errors-warnings-included.log.txt)

These failures look like they have the same root cause and thus I grouped them together.

gsohler, 2025-06-26:
please use -DENABLE_PYTHON during cmake

nomike, 2025-06-27:
Still fails with the same error with commit 24bcd0942a43c8e578074f8be4a1383fa60c8276.

Output from CMake:

```plaintext
source directory: "/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source" (relative from build: "../source")
build directory: "/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/build"
running 'cmake' with arguments ("../source" "-DCMAKE_BUILD_TYPE=RelWithDebInfo" "-DCMAKE_INSTALL_PREFIX=/gnu/store/f5h4dxbbi7pgw1b9igpvyphqs03z62yb-pythonscad-2025.06.26-0.24bcd09" "-DCMAKE_INSTALL_LIBDIR=lib" "-DCMAKE_INSTALL_RPATH_USE_LINK_PATH=TRUE" "-DCMAKE_INSTALL_RPATH=/gnu/store/f5h4dxbbi7pgw1b9igpvyphqs03z62yb-pythonscad-2025.06.26-0.24bcd09/lib" "-DCMAKE_VERBOSE_MAKEFILE=ON" "-DCMAKE_BUILD_TYPE=Release" "-DUSE_BUILTIN_CLIPPER2=OFF" "-DUSE_BUILTIN_MANIFOLD=OFF" "-DUSE_BUILTIN_OPENCSG=OFF" "-DMANIFOLD_PYBIND=OFF" "-DMANIFOLD_TEST=OFF" "-DEXPERIMENTAL=ON" "-DENABLE_PYTHON=ON" "-DOPENSCAD_VERSION=2025.06.26-0.24bcd09" "-DOPENSCAD_COMMIT=24bcd0942a43c8e578074f8be4a1383fa60c8276" "-DPYTHON_VERSION=3.11" "-DENABLE_EGL=ON" "-DENABLE_GLX=ON")
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /gnu/store/a3lsdsalcmg5wnk67869af7wljprkbam-pkg-config-0.29.2/bin/pkg-config (found version "0.29.2") 
-- Eigen: 3.4.0
-- Found Boost: /gnu/store/di4520fl1q7lkxahblsvpd2079zhx0q4-boost-1.83.0/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.61") found components: system regex program_options 
-- Boost: 1.83.0
-- Harfbuzz: 8.3.0
-- Fontconfig: 2.14.0
-- Checking for one of the modules 'glib-2.0'
-- Glib: 2.82.1
-- OPENCSG_DIR: 
-- OpenCSG include found in /gnu/store/29965zfc11j10njskf9lmx8196b05lys-opencsg-1.8.1/include
-- OpenCSG library found in /gnu/store/29965zfc11j10njskf9lmx8196b05lys-opencsg-1.8.1/lib/libopencsg.so
-- OpenCSG: 1.8.1
-- USE_GLAD/USE_GLEW not specified: Defaulting to GLAD
-- OpenGL: /gnu/store/mc0l896lmq74ak7gy55s1b4z68h19440-mesa-25.1.3/lib/libGL.so;/gnu/store/29kjd06xf7140manr0piyzvhx3hj4aq5-glu-9.0.2/lib/libGLU.so
-- Using header-only CGAL
-- Targeting Unix Makefiles
-- Using /gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/bin/c++ compiler.
-- Found GMP: /gnu/store/kka705681m1hq98b9jz98vxk9s5qd4ld-gmp-6.3.0/lib/libgmp.so
-- Found MPFR: /gnu/store/pz4xkjl28w0pqf8nhgv68xix17ap6dsx-mpfr-4.2.1/lib/libmpfr.so
-- Found Boost: /gnu/store/di4520fl1q7lkxahblsvpd2079zhx0q4-boost-1.83.0/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.66")  
-- Boost include dirs: /gnu/store/di4520fl1q7lkxahblsvpd2079zhx0q4-boost-1.83.0/include
-- Boost libraries:    
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Using gcc version 4 or later. Adding -frounding-math
-- CGAL: 5.6.1
-- CGAL: Using target CGAL::CGAL CGAL::CGAL_Core
-- libzip: 1.9.2
-- Freetype: 2.13.0
-- LibXml2: 2.9.14
-- Found Nettle: /gnu/store/al613p11xv5w1xmnqn7ykw0x6d4b0539-nettle-3.9.1/lib/libnettle.so (found suitable version "3.9.1", minimum required is "3.4")
-- Found Python: /gnu/store/s6hivi3vwc1cyyw853y9c8xwahlqym5j-python-3.11.11/include/python3.11 (found suitable exact version "3.11.11") found components: Development Development.Module Development.Embed
-- Python enabled
-- HidAPI: 0.14.0
-- SpNav: found
-- CAIRO_INCLUDEDIR: /gnu/store/gh9cgprk2x0zq7xw44dal0v8s3wixpj3-cairo-1.18.2/include
-- CAIRO_INCLUDE_DIRS: /gnu/store/7ah6i5829f5ha6bdfzcj1gf21115xyyd-freetype-2.13.0/include/freetype2;/gnu/store/zv420d7jscbcwkzndxb9n9dma4xhmf43-pixman-0.42.2/include/pixman-1;/gnu/store/gh9cgprk2x0zq7xw44dal0v8s3wixpj3-cairo-1.18.2/include/cairo;/gnu/store/qj9jhgqn2b9g7yxdhyw6ki1vz241apkd-libpng-1.6.39/include/libpng16
-- CAIRO_LIBRARIES: cairo
-- Cairo: 1.18.2
-- Flex: 2.6.4
-- Found CURL: /gnu/store/cjvaagi44q7r0nhx49r91kmi96xs6qfh-curl-8.6.0/lib/libcurl.so (found version "8.6.0")
-- curl: found in /gnu/store/cjvaagi44q7r0nhx49r91kmi96xs6qfh-curl-8.6.0/include
-- Bison: 3.8.2
-- Found Lib3MF: 3mf;zip;z (found version "2.2.0") 
-- lib3mf: 2.2.0
-- Qt5: 5.15.15
-- QScintilla: 2.13.4
-- DBus input driver enabled
-- Qt5Gamepad input driver enabled
[... 51 lines cut ...]
-- Headless build:        OFF
-- NULLGL build:          OFF
-- OpenGL wrangler:       GLAD
-- Profiling/coverage:    OFF
-- Use mimalloc:          ON
-- Use built-in OpenCSG:  OFF
-- CSG node id-prefix:    OFF
-- Offline documentation: OFF
--  
-- GL Offscreen Context:  Unix EGL
--  
-- Input Drivers
-- - HID API Driver (enabled)
-- - Space Navigator Library Driver (enabled)
-- - Joystick Driver (enabled)
-- - DBus Remote Driver (enabled)
-- - Python support ( enabled )
--  
-- DEFINES = ENABLE_EXPERIMENTAL;_REENTRANT;UNICODE;_UNICODE;_USE_MATH_DEFINES;EIGEN_DONT_ALIGN;BOOST_DLL_USE_STD_FS;ENABLE_OPENCSG;ENABLE_CGAL;CGAL_DEBUG;CGAL_USE_GMPXX;ENABLE_LIBZIP;ENABLE_PYTHON;ENABLE_HIDAPI;ENABLE_SPNAV;ENABLE_CAIRO;ENABLE_LIB3MF;USE_GLAD;USE_ [... cut ...] DGET;ENABLE_DBUS;ENABLE_QGAMEPAD;OPENSCAD_VERSION=2025.06.26-0.24bcd09;OPENSCAD_SHORTVERSION=2025.06.26;OPENSCAD_YEAR=2025;OPENSCAD_MONTH=6;OPENSCAD_DAY=26;OPENSCAD_DAY=26;OPENSCAD_COMMIT=24bcd0942a43c8e578074f8be4a1383fa60c8276;STACKSIZE=8388608;OPENSCAD_OS="Unix";ENABLE_EGL;USE_MANIFOLD_TRIANGULATOR;ENABLE_JOYSTICK;ENABLE_MANIFOLD
-- LINK OPTIONS = -v
--  
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/build
```

nomike, 2025-07-03:
This has been fixed with 3914c3d.

closed 2025-07-07 by nomike