pythonscad/pythonscad #35: Failing test: 2644 - pythonscad_sdf

nomike, 2025-06-27:
**Describe the bug**
Commit: 24bcd0942a43c8e578074f8be4a1383fa60c8276

**ctest output**

```plaintext
    Start 2644: pythonscad_sdf
1/1 Test #2644: pythonscad_sdf ...................***Failed    0.65 sec
sdf
test_cmdline_tool: args:['/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/test_cmdline_tool.py', '--comparator=image_compare', '-c', '/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/build/tests/venv/bin//python', '-s', 'png', '-t', 'pythonscad', '-f', 'sdf', '/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/build/openscad', '/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/data/pythonscad/sdf.py', '--backend=manifold', '--trust-python'] 
test_cmdline_tool: getopt args:['/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/test_cmdline_tool.py', '--comparator=image_compare', '-c', '/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/build/tests/venv/bin//python', '-s', 'png', '-t', 'pythonscad', '-f', 'sdf', '/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/build/openscad', '/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/data/pythonscad/sdf.py', '--backend=manifold', '--trust-python'] 
run_test() cmdline: /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/build/openscad /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/data/pythonscad/sdf.py --backend=manifold --trust-python -o /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/build/tests/output/pythonscad/sdf-actual.png
using font directory: /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/data/ttf
stderr output: Python Code globally trusted
Could not initialize localization (application path is '/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/build').
Traceback (most recent call last):
  File "<string>", line 2, in <module>
ModuleNotFoundError: No module named 'pylibfive'

Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
Geometries in cache: 0
Geometry cache size in bytes: 0
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.071

Image comparison cmdline: /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/build/tests/venv/bin/python ../../source/tests/image_compare.py /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/regression/pythonscad/sdf-expected.png /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/build/tests/output/pythonscad/sdf-actual.png
 actual image: /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/build/tests/output/pythonscad/sdf-actual.png

 expected image: /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/regression/pythonscad/sdf-expected.png

Image comparison return: 1 output: 28.64475202% of 3x3 blocks differ with median block diff: 132.00
Highlight mask saved to: /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/build/tests/output/pythonscad/sdf-actual_mask.png
```

**Images**

Actual: 

[image]

Expected: 

[image]

Highlight Mask: 

[image]

gsohler, 2025-06-27:
pylibfive.py   is avaialale in libraries/python/pylibfive.py
can you check if this is available in your side and is in the path ?

nomike, 2025-06-27:
`libfive` is packaged in guix, but `libfivepy` isn't . I'm packaging that now.

nomike, 2025-06-28:
Screw that. I figured out it's not about packaging that library.

"pylibfive.py"is available in `source/libraries/python/pylibfive.py`.
But it's an out of source build. Could that be the issue?

nomike, 2025-06-29:
Guix per default wants to do an out-of-source build, which PythonSCAD just isn't designed for. I tried to patch the source, adapting for this, but this was more and more tedious, so I now changed the package definition to have "build" be a subdir of "source".

This did not fix this test-case, but it lead to another error output:

```plaintext
          Start 2644: pythonscad_sdf
2644/2677 Test #2644: pythonscad_sdf .................................................***Failed    0.57 sec
sdf
test_cmdline_tool: args:['/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/test_cmdline_tool.py', '--comparator=image_compare', '-c', '/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/build/tests/venv/bin//python', '-s', 'png', '-t', 'pythonscad', '-f', 'sdf', '/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/build/pythonscad', '/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/data/pythonscad/sdf.py', '--backend=manifold', '--trust-python'] 
test_cmdline_tool: getopt args:['/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/test_cmdline_tool.py', '--comparator=image_compare', '-c', '/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/build/tests/venv/bin//python', '-s', 'png', '-t', 'pythonscad', '-f', 'sdf', '/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/build/pythonscad', '/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/data/pythonscad/sdf.py', '--backend=manifold', '--trust-python'] 
run_test() cmdline: /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/build/pythonscad /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/data/pythonscad/sdf.py --backend=manifold --trust-python -o /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/build/tests/output/pythonscad/sdf-actual.png
using font directory: /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/data/ttf
stderr output: Python Code globally trusted
Traceback (most recent call last):
  File "<string>", line 3, in <module>
  File "/tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/libraries/python/pylibfive.py", line 6, in lv_coord
    return lv.x(),lv.y(),lv.z()
           ^^^^
AttributeError: module 'Data' has no attribute 'x'

Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
Geometries in cache: 0
Geometry cache size in bytes: 0
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.051

Image comparison cmdline: /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/build/tests/venv/bin/python ../../tests/image_compare.py /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/regression/pythonscad/sdf-expected.png /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/build/tests/output/pythonscad/sdf-actual.png
 actual image: /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/build/tests/output/pythonscad/sdf-actual.png

 expected image: /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/tests/regression/pythonscad/sdf-expected.png

Image comparison return: 1 output: 28.64475202% of 3x3 blocks differ with median block diff: 132.00
Highlight mask saved to: /tmp/guix-build-pythonscad-2025.06.26-0.24bcd09.drv-0/source/build/tests/output/pythonscad/sdf-actual_mask.png
```

The produced images remained the same.

gsohler, 2025-06-29:
This Looks Like libfive was Not enabled During compilatuon

PR #39 (2025-06-30): Allow unbundling of libfive
This will fix #35 .
closed 2025-06-30 by gsohler