pythonscad/pythonscad #37: Fillet issue

alacritous13, 2025-06-28:
Ran into an issue with the fillet function.

[image]

These airgaps would appear while filleting with a mask, but not when I didn't have a mask.

```
C=cube(10,center=True);
g=cube([5,5,10],center=True).translate([0,0,5])

mm=cube([8,8,3],center=True).translate([0,0,5]);

C=union(g,C)
C=C.fillet(0.5,mm,fn=24)

C.show()
```
I have the exported 3mf/stl but I'm not sure how to post them.

Windows version 2025.06.26

(this bug doesn't affect me or my workflow at this time. I discovered it while familiarizing myself with the programs capacities, and have reported it on the principle of see a bug report a bug)

alacritous13, 2025-06-30:
As soon as a new installer gets built I'll give it a try. I don't have a particular use in mind, but if I've got a tool like this I'm sure opportunities will arise.

The fillet function has a min angle for considering a fillet, right? I saw this in the csg output, but didn't see documentation on it, and was already dealing with other bugs at the time.

gsohler, 2025-06-30:
Hi,

Yes, the min angle parameter is there. But I am not sure if that's the
final way to go.
So there is no documentation about that yet.
My main target is to create valid fillet output in (as-many-as-possible)
cases, but at the same time ,
I am aware that the whole process is quite error prone as it's super easy
to create a case where the fillet algorithm creates invalid output data.
99% of the cases where filleting of an edge goes wrong, if there is NOT at
least "r" space on both sides to form the rounding . And thats very easy to
create by arbitrary CSG operations of cylinders where there is no control ,
where the edges are created.

For that reason I am planning another improvement to prepare the input
edges of the solid , until all planned fillets are possible.

Yes, I will notify, once testing can continue.

alacritous13, 2025-06-30:
Love to hear it. I also saw that Union and Difference had a radius input for fillet documented, but I couldn't get it to work (I didn't get very hard). I'm under the assumption this is in the works but is currently disabled.

(apologies, I am just using this issue to ask all my fillet questions)

gsohler, 2025-07-01:
you can get filleting for union work if you got manifold enabled(only manifold codepath is coded)

however, you might experience some odd filetting yet ...

gsohler, 2025-07-04:
Today independency Release is online which should have your filletting issue fixed

alacritous13, 2025-07-07:
Mesh is sealed, having new issues.

Let me preface this with, I consider fillet to be an experimental and unstable feature. I have no expectations about its capacities, or that any issues with it will be resolved in a timely manner. That said:

The size of a fillet on a rounded part is inconsistent and appears based on the faceting of said circle.

[image]

These are the same shape with different `fn` values for the starting cylinders, and they have significantly different fillets as a result.

```
from openscad import *

fn=32;

C1=cylinder(d=10,h=10,center=False)
C2=cylinder(d=5,h=10,center=True).translate([0,0,10])

FnL=5;
C1a=cylinder(d=10,h=10,center=False,fn=FnL)
C2a=cylinder(d=5,h=10,center=True,fn=FnL).translate([0,0,10])
examp0=difference(C1a,C2a,r=0.5,fn=24)
examp0=examp0.render()
examp0=examp0.translate([15*0,0,0])

examp1=difference(C1,C2,r=0.5,fn=24)
examp1=examp1.render()
examp1=examp1.translate([15*1,0,0])

examp2=difference(C1,C2)
examp2=examp2.fillet(r=0.5)
examp2=examp2.translate([15*2,0,0])

FnL=24;
C1b=cylinder(d=10-0.5*2,h=10,center=False,fn=FnL)
C2b=cylinder(d=5+0.5*2,h=10,center=True,fn=FnL).translate([0,0,10])
examp3=difference(C1b,C2b)
examp3=examp3.render()
SS=sphere(r=0.5,fn=24);
examp3=minkowski(examp3,SS)
examp3=examp3.translate([15*3,0,0])

R1=square([10,10],center=True).translate([0,5]).linear_extrude(0.001,center=True);
R2=square([5,10],center=True).translate([0,10]).linear_extrude(0.001,center=True)
examp4=difference(R1,R2,r=0.5,fn=24);
examp4=examp4.projection()
examp4=examp4.difference(square(200,center=True).translate([-100,0]))
examp4=examp4.rotate_extrude();
examp4=examp4.render()
examp4=examp4.translate([15*4,0,0])

show([examp0,examp1,examp2,examp3,examp4])
```

Here are several similar parts that have been filleted in significantly different ways. `examp4` is what I expected the outcome to look like.

PS: Should I be opening a new issue for this?

gsohler, 2025-07-07:
This is a tough one. But ademittedly absolutely required

[image]
here is 1st progress

gsohler, 2025-07-08:
sorry, 2D is not (yet) supported for fillets. I think I should explicitely disable that feature and throw an error.

Haha, but I am working on an improved version of the fillet which combines edges so long until all fillets get enough space on both sides.
Yeah, i am happy to create another version, once the import bug is understood.

alacritous13, 2025-07-12:
As of Version 2025.07.11, this looks to have been fixed.