pythonscad/pythonscad #40: No GUI on guix build

nomike, 2025-07-03:
**Describe the bug**
I have built 3914c3d on guix. Most of the tests run fine (there are a few left I'm currently working on).

However, I was trying to reproduce an issue and wanted to use pythonscad interactively, but it does not launch a GUI.

When I run it like this from the build directory, I'm getting two lines of log output and that's it:

```plaintext
./pythonscad  ~nomike/coding/3d\ prints/my-designs/Mitre\ Box/Mitre\ Box.scad
Loaded design '/home/nomike/coding/3d prints/my-designs/Mitre Box/Mitre Box.scad'.
This plugin does not support propagateSizeHints()
```

and then it hangs forever.

**To Reproduce**
Steps to reproduce the behavior:

1. Build guix with this package definition, using the `--keep-failed` option on `guix build`:

```scheme
(define-public pythonscad
  (let ((commit "3914c3d0973d92bd5e2fe2be47c1fc568cb05cbb")
        (version "2025.07.03")
        (revision "0"))
    (package
      (name "pythonscad")
      (version (git-version version revision commit))
      (source
       (origin
         (method git-fetch)
         (uri (git-reference
               (url "https://github.com/pythonscad/pythonscad")
               (commit commit)
               ;; Needed for libraries/MCAD, a library specific to OpenSCAD
               ;; which is included as a submodule. All other libraries are
               ;; deleted in the patch-source build phase.
               (recursive? #t)))
         (sha256
          (base32 "1avbddxmd07nn5swh0la3mx6v3xj1v4fcfgxhrr8aqn76anx76h9"))
         (patches (search-patches
                   ;;  "openscad-fix-path-in-expected-test-results-to-acommodate-diff.patch"
                   ;;  "pythonscad-2025.06.25-fix-test-paths.patch"
                   ;; "pythonscad-2025.06.13-Fix-Wrong-filename-in-translation-update.sh.patch"
                   "pythonscad-2025.06.13-fix-glew.patch"
                  ;  "pythonscad-2025.06.26-fix-test-tool.patch"
                   ))
         (file-name (git-file-name name version))))
      (build-system qt-build-system)
      (arguments
       (list
        #:out-of-source? #f
        #:configure-flags
        #~(list "-DCMAKE_BUILD_TYPE=Release"
                "-DUSE_BUILTIN_CLIPPER2=OFF"
                "-DUSE_BUILTIN_MANIFOLD=OFF"
                "-DUSE_BUILTIN_OPENCSG=OFF"
                "-DMANIFOLD_PYBIND=OFF"
                "-DMANIFOLD_TEST=OFF"
                "-DEXPERIMENTAL=ON"
                "-DENABLE_PYTHON=ON"
                (string-append "-DOPENSCAD_VERSION="
                               #$version)
                (string-append "-DOPENSCAD_COMMIT="
                               #$commit)
                (string-append "-DPYTHON_VERSION="
                               #$(version-major+minor (package-version python)))
                "-DENABLE_EGL=ON"
                "-DENABLE_GLX=ON"
                "-DENABLE_LIBFIVE=ON"
[... 277 lines cut ...]
  /gnu/store/fixj7jfkisczxaibjhlnc763vqp5x3hd-libxext-1.3.6/share/fonts
  /gnu/store/nw27myjl7crh9rallpqx27wdlmf54qn4-libxcb-1.17.0/share/fonts
  /gnu/store/x1mms0gyjks84665606bcjsnzri3mr5k-libxdmcp-1.1.5/share/fonts
  /gnu/store/gvax5yl1b2i83lw5ak6r0gi28m4mxqqa-libxau-1.0.12/share/fonts
  /tmp/.fonts
  /tmp/.guix-home/profile/share/fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /tmp/Library/Fonts
  /gnu/store/mjg1gq1w6csggjhy5wh6njyzqpic4avq-font-dejavu-2.37/share/fonts/truetype

GL context creator: EGL (new)
EGL version: 1.5
PNG generator: lodepng

GLAD version: 2.0.4
OpenGL Version: 4.6 (Compatibility Profile) Mesa 25.1.3
GL Renderer: AMD Radeon Graphics (radeonsi, renoir, ACO, DRM 3.61, 6.14.0-22-generic)
GL Vendor: AMD
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

gsohler, 2025-07-03:
Not sure, if the error message is related to the hang
you might try compiling with ENABLE_PYTHON=0 to see if the hang is related
to python initialization

also watch out this one

https://forum.qt.io/topic/150802/qt5-error-using-geant4-this-plugin-does-not-support-propagatesizehints/2

nomike, 2025-07-03:
From what I can deduct from that thread ad qt.io the message on the console doesn't seem to be related.

I'm trying a build with `-DENABLE_PYTHON=0` now. Let's see where this leads us to.

nomike, 2025-07-07:
I figured something out:

During the build guix is controlling the environment.
It set's the values like this:

[environment-variables.gz](https://github.com/user-attachments/files/21094299/environment-variables.gz)

When I run pythonscad like that, it will not show the GUI.
However, when I run it directly from a native ubuntu shell without modifying the environment, it will show the GUI as expected.

It looks like this needs some more investigation on my end.

nomike, 2025-07-07:
And I figured it out:

It is this environment variable: `export QT_QPA_PLATFORM="offscreen"`.
So I guess we can ignore that.

I will close this issue now.