pythonscad/pythonscad #44: Can't import packages/other py files

alacritous13, 2025-07-07:
I have file `Ted.py`

```
from openscad import *

def Box(a=1,b=[0,0,0]):#{
    if not(isinstance(a,list)):#{
        a=[a,a,a];
    #}
    c=cube(a,center=True);
    c=c.translate([a[0]*b[0]/2,a[1]*b[1]/2,a[2]*b[2]/2]);
    c.origin=[[1,0,0,0],[0,1,0,0],[0,0,1,0],[0,0,0,1]];
    return c;
#}
```
And I have another file next to it in my directory.
```
from openscad import *
from Ted import *

Box(3).show()
```

It's not working on Windows version (2025.07.04) and produces this error:
```
Parsing design (AST generation)...
Running Python 3.12.9 without venv.
ERROR: Traceback (most recent call last):
  File "<string>", line 2, in <module>
ModuleNotFoundError: No module named 'Ted'

Execution aborted
```

It does work on 2025.06.26.

I've seen other discrepancies that make it look like it can occasionally work with a package. It doesn't work with my install of numpy, but it does with forbbidenfruit. I'm including the code relevant to that, but most of it was stollen from reddit and I don't know how it works.

```
def load_numpy():#{
    # My local windows Python 3.11 pip3 site-packages location.
    # You would need to separately install the official Python 3.11 runtime.
    import os
    login = os.getlogin()
    WINDOWS_PY_311_PIP_LIBS = f"C:\\Users\\liebe\\AppData\\Local\\Programs\\Python\\Python311\\Lib\\site-packages"
    
    # I already had pip3 install numpy out-of-band on my Windows machine after installing Python 3.11 runtime.
    
    # Add Py 3.11 site-packages to path.
    if WINDOWS_PY_311_PIP_LIBS not in sys.path:#{
        sys.path.append(WINDOWS_PY_311_PIP_LIBS)
    #}
#}
try:#{
    load_numpy();
#}
except:#{
    print("Didn't initalize numpy")
#}
from openscad import *
import numpy as np
from forbiddenfruit import curse, reverse
```

gsohler, 2025-07-07:
error reproduced in windows only.

gsohler, 2025-07-07:
import sys
import os
sys.path.append(os.path.dirname(modelpath()))    
from Ted import *

this is a quickfix

alacritous13, 2025-07-08:
Works to be able to use adjacent files, doesn't fix the issue with Numpy.

Getting this error message:
```
arsing design (AST generation)...
Running Python 3.12.9 without venv.
ERROR: Traceback (most recent call last):
  File "C:\Users\liebe\AppData\Local\Programs\Python\Python311\Lib\site-packages\numpy\_core\__init__.py", line 23, in <module>
    from . import multiarray
  File "C:\Users\liebe\AppData\Local\Programs\Python\Python311\Lib\site-packages\numpy\_core\multiarray.py", line 10, in <module>
    from . import overrides
  File "C:\Users\liebe\AppData\Local\Programs\Python\Python311\Lib\site-packages\numpy\_core\overrides.py", line 7, in <module>
    from numpy._core._multiarray_umath import (
ModuleNotFoundError: No module named 'numpy._core._multiarray_umath'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\liebe\AppData\Local\Programs\Python\Python311\Lib\site-packages\numpy\__init__.py", line 127, in <module>
    from numpy.__config__ import show_config
  File "C:\Users\liebe\AppData\Local\Programs\Python\Python311\Lib\site-packages\numpy\__config__.py", line 4, in <module>
    from numpy._core._multiarray_umath import (
  File "C:\Users\liebe\AppData\Local\Programs\Python\Python311\Lib\site-packages\numpy\_core\__init__.py", line 49, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.12 from "C:/Program Files/PythonSCAD/openscad-python"
  * The NumPy version is: "2.2.6"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: No module named 'numpy._core._multiarray_umath'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<string>", line 15, in <module>
  File "C:\Users\liebe\AppData\Local\Programs\Python\Python311\Lib\site-packages\numpy\__init__.py", line 132, in <module>
    raise ImportError(msg) from e
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.

Execution aborted
 ```

Running this:
```
from openscad import *
import sys
import os
sys.path.append(os.path.dirname(modelpath()))

login = os.getlogin()
WINDOWS_PY_311_PIP_LIBS = f"C:\\Users\\liebe\\AppData\\Local\\Programs\\Python\\Python311\\Lib\\site-packages"
if WINDOWS_PY_311_PIP_LIBS not in sys.path:#{
    sys.path.append(WINDOWS_PY_311_PIP_LIBS)
#}

from Ted import *
import numpy as np
from forbiddenfruit import curse, reverse

Box(3).show()
```

gsohler, 2025-07-08:
No code changes since 2025-06-26 in the relevant locatoin, but Now I realize, what happened.
Since some of my source files are not downloadable anymore, I upgraded to python3.12 for windows(now i remember, can see it in your logs)
path setting appears to work differently now.
I hope to get it working again.

gsohler, 2025-07-08:
sorry, python312._pth

alacritous13, 2025-07-08:
It did not fix it. But I also don't really understand the pathing stuff I've been doing to make it work, so I might be shooting myself in the foot.

I still have Python11, do I also need to update? (the error message makes it look like numpy might be version specific)

gsohler, 2025-07-09:
> I've got the packages installed under a 3.12 version. Along with the quick
> fix, I'm now able to access adjacent files and all of the libraries I use.

Thank you for your feedback!

Many python packages (numpy also )have a shared library portion which
matches a certain python version.
Therefore you had to install numpy again on 3.12

If should be possible to do everything *without* the quickfix.
Did you make sure, that the python312._pth in the pythonscad install
directory is deleted ?
what is the content of sys.path before the quickfix ?

alacritous13, 2025-07-12:
As of 2025.07.11: I can import from local files without issues. I have to jump through some hoops to get other packages working (numpy) but I always had to do that, so unless you disagree I'm considering this as fixed.