pythonscad/pythonscad #48: Wrap first and last segment misaligned

alacritous13, 2025-07-11:
While wrapping a shaping using the `wrap` function, the first segment is rotated such that it is inline with the second segment, and that the last segment is point towards the center.

[image]

[image]

[image]

```
from openscad import *

N=2
rR=3.7
fn=13

#g=cube([rR*3,0.2,2])
g=text("Hello world"*N).linear_extrude(0.5).rotate([90,0,0]).roty(-2)
g=g.wrap(rR)
g.show()
circle(r=rR).color('green').show()
```

As a bonus issue, `wrap` didn't like it when I tried specifying radius or diameter (`wrap(r=val)` and `wrap(d=val)`) and errored out.

Windows 2025.07.04

alacritous13, 2025-07-12:
Yes, that would be a lot more logical

gsohler, 2025-07-12:
yep, i will shedule that for next release

alacritous13, 2025-08-10:
It's working.

A quirk, the UI View All zooms out as though the part is much bigger than it is, picture included showing how far it zooms out after I've pressed the button. Odd, but not disruptive.

[image]

closed 2025-08-25 by alacritous13