pythonscad/pythonscad #49: Failing test: Failing test:

nomike, 2025-07-12:
**Describe the bug**
Commit: b55beb1f22fb1663938fc1aa9aec61821090cd1f
The test did not fail with bfe161206ac65a9afcc461086976de9dcd3832a9

**ctest output**

```plaintext
1/5 Test  #964: preview-cgal_issue267-normalization-crash ..............***Failed    3.07 sec
issue267-normalization-crash
test_cmdline_tool: args:['/tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/tests/test_cmdline_tool.py', '--comparator=image_compare', '-c', '/tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/build/tests/venv/bin//python', '-s', 'png', '-e', 'preview', '-t', 'preview-cgal', '-f', 'issue267-normalization-crash', '/tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/build/pythonscad', '/tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/tests/data/scad/3D/issues/issue267-normalization-crash.scad', '--backend=cgal'] 
test_cmdline_tool: getopt args:['/tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/tests/test_cmdline_tool.py', '--comparator=image_compare', '-c', '/tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/build/tests/venv/bin//python', '-s', 'png', '-e', 'preview', '-t', 'preview-cgal', '-f', 'issue267-normalization-crash', '/tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/build/pythonscad', '/tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/tests/data/scad/3D/issues/issue267-normalization-crash.scad', '--backend=cgal'] 
run_test() cmdline: /tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/build/pythonscad /tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/tests/data/scad/3D/issues/issue267-normalization-crash.scad --backend=cgal -o /tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/build/tests/output/preview-cgal/issue267-normalization-crash-actual.png
using font directory: /tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/tests/data/ttf
stderr output: WARNING: undefined operation (bool > number) in file issue267-normalization-crash.scad, line 23
WARNING: undefined operation (bool > number) in file issue267-normalization-crash.scad, line 23
WARNING: undefined operation (bool > number) in file issue267-normalization-crash.scad, line 23
WARNING: undefined operation (bool > number) in file issue267-normalization-crash.scad, line 23
WARNING: undefined operation (bool > number) in file issue267-normalization-crash.scad, line 23
Compiling design (CSG Products normalization)...
Normalized CSG tree has 3073 elements
Geometries in cache: 7
Geometry cache size in bytes: 82632
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:02.418

Image comparison cmdline: /tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/build/tests/venv/bin/python ../../tests/image_compare.py /tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/tests/regression/preview/issue267-normalization-crash-expected.png /tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/build/tests/output/preview-cgal/issue267-normalization-crash-actual.png
 actual image: /tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/build/tests/output/preview-cgal/issue267-normalization-crash-actual.png

 expected image: /tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/tests/regression/preview/issue267-normalization-crash-expected.png

Image comparison return: 1 output: 0.28770986% of 3x3 blocks differ with median block diff: 47.74
Highlight mask saved to: /tmp/guix-build-pythonscad-2025.07.11-0.b55beb1.drv-0/source/build/tests/output/preview-cgal/issue267-normalization-crash-actual_mask.png
```

**Images**

Actual:

[image]

Expected:

[image]

Highlight mask:

[image]

gsohler, 2025-07-12:
yours is working whereas here its crashing.
in thart case just recreate the image and you will be pass. fine with me

nomike, 2025-07-20:
As the goal of this test is to confirm that PythonSCAD is not crashing and not about the actual image produced. I re-generated the image to make the test succeed.

Has been fixed in 3f476af74e7478859086a7a5a041953749cd8de1.

closed 2025-07-20 by nomike