pythonscad/pythonscad #51: Readme gives wrong instructions

bassklampfe, 2025-07-12:
[pythonscad/README.md](https://github.com/pythonscad/pythonscad/README.md) states to use command
`git clone https://github.com/pythohn/pythonscad.git`
but the right command is
`git clone https://github.com/pythonscad/pythonscad.git`

Then the next commands listed are
```
git submodule update --init --recursive
cd pythonscad
git submodule update --init --recursive
```
The first line is wrong, before `cd pythonscad` git commands will not work for the repository. Remove first line!

```
# make sure to get cryptopp and python dev packages installed, additionally
```
this is not very helpful. Skipping this

```
cmake -DEXPERIMENTAL=1 -DENABLE_PYTHON=1 -DENABLE_LIBFIVE=1 ..
```
results in
```
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- 
-- No build type selected, default to: Release
-- Override standard malloc (MI_OVERRIDE=ON)
-- Use the C++ compiler to compile (MI_USE_CXX=ON)
-- 
-- Library base name: mimalloc
-- Version          : 1.7
-- Build type       : release
-- C++ Compiler     : /usr/bin/c++
-- Compiler flags   : -Wall;-Wextra;-Wno-unknown-pragmas;-fvisibility=hidden;-ftls-model=initial-exec;-fno-builtin-malloc
-- Compiler defines : 
-- Link libraries   : /usr/lib/x86_64-linux-gnu/libpthread.a;/usr/lib/x86_64-linux-gnu/librt.a
-- Build targets    : static
-- 
-- Eigen: 3.4.0
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.61") found components: system regex program_options 
-- Boost: 1.74.0
-- Harfbuzz: 2.7.4
-- Fontconfig: 2.13.1
-- Checking for one of the modules 'glib-2.0'
-- Glib: 2.72.4
-- OPENCSG_DIR: 
-- OpenCSG include found in /usr/include
-- OpenCSG library found in /usr/lib/x86_64-linux-gnu/libopencsg.so
-- OpenCSG: 1.4.2
-- USE_GLAD/USE_GLEW not specified: Defaulting to glew
-- GLEW: 2.2.0
-- OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so;/usr/lib/x86_64-linux-gnu/libGLX.so;/usr/lib/x86_64-linux-gnu/libGLU.so
CMake Warning at /usr/lib/x86_64-linux-gnu/cmake/CGAL/CGALConfig.cmake:92 (message):
  CGAL_DATA_DIR cannot be deduced, set the variable CGAL_DATA_DIR to set the
  default value of CGAL::data_file_path()
Call Stack (most recent call first):
  CMakeLists.txt:434 (find_package)

[... 161 lines cut ...]
--  
-- GL Offscreen Context:  Unix EGL GLX
--  
-- Input Drivers
-- - HID API Driver (disabled)
-- - Space Navigator Library Driver (disabled)
-- - Joystick Driver (enabled)
-- - DBus Remote Driver (enabled)
-- - Python support ( enabled )
--  
-- DEFINES = ENABLE_EXPERIMENTAL;_REENTRANT;UNICODE;_UNICODE;_USE_MATH_DEFINES;USE_MIMALLOC;MI_OVERRIDE;MI_LINK_STATIC;EIGEN_DONT_ALIGN;BOOST_DLL_USE_STD_FS;ENABLE_OPENCSG;OPENCSG_GLEW;BOOST_BIND_GLOBAL_PLACEHOLDERS;ENABLE_CGAL;CGAL_DEBUG;CGAL_USE_GMPXX;ENABLE_LIBZIP;ENABLE_LIBFIVE;ENABLE_PYTHON;ENABLE_CAIRO;USE_GLEW;USE_QOPENGLWIDGET;ENABLE_DBUS;OPENSCAD_VERSION=2025.07.12;OPENSCAD_SHORTVERSION=2025.07.12;OPENSCAD_YEAR=2025;OPENSCAD_MONTH=7;OPENSCAD_DAY=12;OPENSCAD_DAY=12;STACKSIZE=8388608;OPENSCAD_OS="Unix";ENABLE_EGL;GLEW_EGL;ENABLE_GLX;USE_MANIFOLD_TRIANGULATOR;ENABLE_JOYSTICK;ENABLE_MANIFOLD
-- LINK OPTIONS = -v
--  
-- Configuring incomplete, errors occurred!

```

My system :
```
PRETTY_NAME="Ubuntu 22.04.5 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.5 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
```

bassklampfe, 2025-07-12:
` luaL_register` : Ok, you're on Lua5.4. Solution found here: https://github.com/franko/graph-toolkit/issues/4

```
-    luaL_register (L, NULL, OpenSCADObject_mt);
+    luaL_setfuncs (L, OpenSCADObject_mt, NULL);
```

gsohler, 2025-07-12:
> Bug in CMakeText.lst
>
> 	  if (${PYTHON_VERSION}  STREQUAL "OFF")
>       find_package(Python  REQUIRED COMPONENTS Development)
>     else()
>       find_package(Python  ${PYTHON_VERSION} EXACT REQUIRED COMPONENTS Development)
>     endif()
>     find_package(Python  ${PYTHON_VERSION} EXACT REQUIRED COMPONENTS Development)
>
> The last line is wrong (copy-n-paste error?), remove please!
>
> after fixing this, and installing guile-3.0-dev
>
> cmake -DEXPERIMENTAL=1 -DENABLE_PYTHON=1 -DENABLE_LIBFIVE=1 ..
>
> finishes successful and then also make -j$(nproc)
>
> Next step: Switch to branch lua (but not today, I'm busy for the rest of
> the day)

great!
guile ist not needed. it's just optional for libfive. do dont have it either

closed 2025-07-14 by bassklampfe

bassklampfe, 2025-07-18:
I can look in my old code for some examples. Please be patient...

bassklampfe, 2025-07-20:
@gsohler : Ok, here are some examples.
I could not use my real code, because it contains proprietary information, but I believe these examples are good enough to give you some clues.

**Case 1:** 
You known exactly how many arguments of which type you expect:

```C++
static int f_number_number(lua_State *L)
{
	// evaluate arguments
	if(lua_gettop(L)!=2) luaL_error(L,"expect 2 arguments");
	double a1=luaL_checknumber(L,1);
	double a2=luaL_checknumber(L,2);
	// use arguments
	lua_pushnumber(L,sqrt(a1*a1+a2*a2));
	return 1;
}
```

Usage:
```lua
print(luaargs.f_number_number(3,4))
```
Result
```
5
```

**Case 2:** 
You expect a variable list of (unnamed) arguments, identified by type and/or position in the argument list
```C++
static int f_variable_args(lua_State *L)
{
	// preset with default values
	int int_arg=-1;
	const char*str_arg_1=nullptr;
	const char*str_arg_2=nullptr;
	// loop over number of arguments
	int top=lua_gettop(L);
	for(int i=1; i<=top; i++)	//  all args
	{
		int t=lua_type(L, i);	// get the type
		switch(t)
		{
		case LUA_TSTRING:  /* strings */
			if(str_arg_1==nullptr)
				str_arg_1=lua_tostring(L, i);
			else if(str_arg_2==nullptr)
				str_arg_2=lua_tostring(L, i);
			else
				luaL_error(L,"str_arg_1 and str_arg_2 already given");
			break;
		case LUA_TNUMBER:  /* numbers */
			if(int_arg>=0)
				luaL_error(L,"int_arg already given");
			int_arg=lua_tonumber(L, i);
			if(int_arg<0||int_arg>0x7F)
				return luaL_error(L,"invalid int_arg");
			break;
		default:  /* other values */
			return luaL_error(L,"invalid argument");
		}
	}

	// check arguments
	if(int_arg<0)
		return luaL_error(L,"no int_arg given");
	if(str_arg_1==nullptr)
		return luaL_error(L,"no str_arg_1 given");
	if(str_arg_2==nullptr)
		return luaL_error(L,"no str_arg_2 given");
	// further processing ...

	luaL_Buffer b;
	luaL_buffinit(L, &b);
	for(int i=1;i<=int_arg;++i)
	{
		luaL_addstring(&b,str_arg_1);
		luaL_addstring(&b,str_arg_2);
	}
	luaL_pushresult(&b);
    return 1;
}
```
Usage:
```lua
print(f_variable_args(3,"a","b"))
```
Result:
```
ababab
```

**Case 3:** 
Table with named argumentsargument list
```C++

static int  f_table_with_named_args(lua_State *L)
{
	// default values
	double x=0;
	double y=0;
	double w=1;
	double h=1;
	const char*valign=nullptr;
	const char*halign=nullptr;

	// check arguments
	enum { ARG_TABLE=1};
	if(lua_gettop(L)!=1 || lua_type(L,ARG_TABLE)!=LUA_TTABLE)
		luaL_error(L,"expect 1 table as argument");

	// iterate over the table elements
	for(lua_pushnil(L);lua_next(L,ARG_TABLE);lua_pop(L,1))
	{
		#define T_KEY -2
		#define T_VALUE -1
		const char *arg_key=lua_tostring(L,T_KEY);
		if(!arg_key)
			luaL_error(L,"argument name must be string");
		int arg_typ=lua_type(L,T_VALUE);
		if(arg_typ==LUA_TNUMBER)
		{
			if(strcmp(arg_key,"x")==0)
				x=lua_tonumber(L,T_VALUE);
			else if(strcmp(arg_key,"y")==0)
				y=lua_tonumber(L,T_VALUE);
			else if(strcmp(arg_key,"w")==0)
				w=lua_tonumber(L,T_VALUE);
			else if(strcmp(arg_key,"h")==0)
				h=lua_tonumber(L,T_VALUE);
			else
				luaL_error(L,"unknown argument name for number");
		}
		else if(arg_typ==LUA_TSTRING)
		{
			if(strcmp(arg_key,"valign")==0)
				valign=lua_tostring(L,T_VALUE);
			else if(strcmp(arg_key,"halign")==0)
				halign=lua_tostring(L,T_VALUE);
			else
				luaL_error(L,"unknown argument name for string");
		}
		else
		{
			luaL_error(L,"argument not string or number");
		}
	}
	std::cout<<"x="<<x<<"\n";
	std::cout<<"y="<<y<<"\n";
	std::cout<<"w="<<w<<"\n";
	std::cout<<"h="<<h<<"\n";
	std::cout<<"valign="<<(valign?:"nullptr")<<"\n";
	std::cout<<"halign="<<(halign?:"nullptr")<<"\n";
	return 0;
}
```
Usage:
```Lua
f_table_with_named_args{}
f_table_with_named_args{x=123,y=345,w=-1,valign="left"}
```
Result
```
x=0
y=0
w=1
h=1
valign=nullptr
halign=nullptr

x=123
y=345
w=-1
h=1
valign=left
halign=nullptr
```

bassklampfe, 2025-07-20:
Some caveats:

**Typeconversion:**
If you have a number argument and call `lua_tostring(…)` without checking `lua_type(…)` before, the number argument (!) will be silently converted to string
Same: If you have a string argument and call `lua_tonumber(…)` without checking `lua_type(…)` before, the string argument (!) will be silently converted to number (if possible)

**Strings:**
`const char*str=lua_tostring(…)` is valid, as long as you stay in the scope, where lua_tostring was called. And it is performant because of zero-copy.

If you want the value to stay alive beyond the scope then you have to store the value (e.g. in std::string or QString)

**Numbers:**
Lua5.1 does only know 'number' as type, which is good enough for double and int (as long as the int value does not exceed 52 bits).

I know, Lua5.4 can distinguish integer and float values, but I have no experience in this. Please read the docs 🤓

gsohler, 2025-07-28:
Sorry, Just now I got time to look into your lua examples, and yes:
they are *very* comprehensive - I could use them right away to extend my

int LuArg_ParseTupleAndKeywords(lua_State *L, const char *fmt, char **kwlist , ...)

function. 
PyArg_ParseTupleAndKeywords functin in python can be used to parse a single  function command  with positional and optional named arguments.
Each argument type is specified as a letter in "fmt" parameter. 
Especially "O" means : arbritary Python argument, will be typechecked and parsed later

With your great example its an easy task to parse a FLAT argument list becasue I can easily work on the L stack provided , but as soon the arguments are hierarchical, I encounter huge issues. Imagine i'd like to parse the coordinates of a polyhedron point list, i immediately see 3 levels of hierarchy which i have to parse in 3 different steps.
Ideally sub-steps can work on "virtual stacks" such the functions can be reused

For the polyhedron ideally parsing works like this

toplevel parsing parses polyhedron function and returns substack for "points" argument
next parsing parsess points as a list and returns substack for single coordinates
next parsing step parses coordinate as a list and parses x,y,z value

I have seen the lua_geti function which can accept indices for two different stack indices, but its not very flexible once hierarchy exceeds 2..

Is there a versatile solution possible in lua ?