pythonscad/pythonscad #52: Skin-non repeatable

alacritous13, 2025-07-13:
Was testing out skin trying to figure out how it did calculations because I've so far not see it working reliably enough for me to start using it, and realized it wasn't repeatable for what should have effectively been the same shape (so hopefully the issues I'm seeing are an indexing bug and not a fundamental issue related to trying to extrapolated between two concave shapes).

[image]

```
from openscad import *

shap = square([5,5],center=True) | square([2,7],center=True);

shap1 = shap.translate([0,0,0]);

shap2a=shap.translate([0,0,5]).rotate([0,0,12])
shap2b=shap.rotate([0,0,12]).translate([0,0,5])

skin(shap,shap2a).show()
skin(shap,shap2b).translate([7,0,0]).show()
```

Additionally, if I could get a quick rundown of what the logic is suppose to be, I'd be interested in that. If I had to guess, for non-concave shapes, its suppose to produce the same results that you would see with the hull function (if the 2d shapes were extruded a miniscule amount), but it looks like the index occasionally gets rotated. No idea what logic is being used to get concave shapes to work.

[image]
Same code as above, but `shap` was hulled initially, and a traditional SCAD hull example was included on the right.

gsohler, 2025-07-13:
Sorry, i will be able to Help you mich with skin().
Its  the Feature of someone Else. But i am happy to fix, when you Spot
Errors in the Code.

alacritous13, 2025-07-14:
Did it by chance come from bols2? I just tried out the skin function there, and while I can't quite get the same bug, I did getting some weirdness.

[image]
Yes, those are zero thickness panels.

I'd be interested in playing with my own method. Necessary to my skill base, it would have to be before manifold, not integrated into it. My first hurdle, I'm not sure how to get 2d shapes to return a list of points in 3d. Skin involves positioning 2d shapes in 3d, and as such needs the coordinates in 3d. Not sure if this just isn't possible, but I'm sure if their isn't an easy way, there's a hard way I can find.

alacritous13, 2025-07-14:
I think I did extrude a 2.5 already, didn't even notice it was so organic. I've seen the documentation about 1D shapes, but couldn't quite figure them out initially and didn't dig further (yet). My issue with `mesh`, when I've tried it with 2.5d shapes, it outputs only 2 coordinates, with a 2d shape I can go and just add zeros to each point, but for a 2.5 shape, it has actual z coordinates, it just isn't returning them. I didn't test it particularly much, but I think the output was effectively what I would expect from projection (I think that doesn't work on 2.5d shapes, might be a cool feature if it isn't already implemented (I say as if I've used projection more than twice outside of testing)).

Thanks for being so responsive to my feedback. I can understand the frustration of trying to work with people who don't understand the codebase they're commenting on.

gsohler, 2025-07-14:
For 2D coordinates, mesh should actually return a list of  all 2D points ,
it should be more than 2 points to form a triangle

print(square(4).mesh())
yields

[[[0.0, 0.0], [4.0, 0.0], [4.0, 4.0], [0.0, 4.0]]]

issue with the 2D stuff, is that its *very* virtual and only applies the
2.5D stuff in the very last moment

src/geometry/Polygon2d.h declares a line

Transform3d trans3d;

which actually stores the 2.5D information

alacritous13, 2025-07-15:
[image]
What I was trying to convey is that it only gives me the `[x,y]` component, resulting in the points being flattened. The above image is a square that's been rotated, the used mesh to get the points and placed a pink cube at each.

Mesh returns `[[[-2.5, -1.7677669529663689], [2.5, -1.7677669529663689], [2.5, 1.7677669529663689], [-2.5, 1.7677669529663689]]]`

```
from openscad import *

c=square(5,center=True).rotate(45,[1,0,0]);

P=cube(0.5,center=True).color("pink");

L=c.mesh();

c.show()

print(L)
for x in L[0]:#{
    P.translate(x).show()
#}
```

This doesn't seam to be a particularly concerning thing to fix. I can think of use cases, but none of them are of much importance.