pythonscad/pythonscad #53: Linear_extrude with equation not working when number of points change mid extrude

alacritous13, 2025-07-15:
[image]

```
def CCress(h=0):#{
    H=5;
    if h<3:#{
        q=4
    #}
    else:#{
        q=5
    #}
    return circle(5,fn=q).rotz(h/H*90).mesh()[0]
#}

linear_extrude(CCress,height=5,fn=5).show()
```

Above function produces the above image. Shape looks fine up until it hits the height 3 where it adds an extra point to the shape. It's hard to tell, but it kind of looks like it decided point 5 (of the first layer with 5 points) was point 1 of the next layer. This shape is water tight.

Then, just in general, the entire process has caching issues. All math is done and stored in the cache, making exported csg trees useless (bad form, but it works). The shapes are not re-rended when the formulas are updated, require manually clearing the cache (they cache a separate value if you change the fn or height). And the caches are assigned in order of being called, so if multiple calls have the same height and fn value, they might switch which is accessing which cache if you add or remove one during editing (clearing the cache fixes this). Long story short clearing the cache after every change keeps things working.

Windows, version 2025.07.11

gsohler, 2025-07-15:
For me updating the forumula is working without clearing the cache,.
https://github.com/pythonscad/pythonscad/blob/master/src/core/LinearExtrudeNode.cc#L157
the key of the LinearExtrdueNode has a different key each time due to the rand() function.
Its stupid imperformant, but it should work good enough for the time beeing.

gsohler, 2025-07-16:
> Yeah, I wrote the bug report and have been thinking about this all of last
> night. There's no good way to connect segments of different point count, it
> should be checked and produce a warning if done. But any way to actually
> connect them is too arbitrary.
>
> I don't know what to tell you about the cache. I can see the key being
> randomized, but they still won't update till the cache is cleared. It's an
> annoyance, but the thing technically works so I can't complain to much. I'm
> bugged that we get a random can to a key value in the csg, and not a
> massive call of polyhedron, but once again, it works.

Hi,

There *is* a way to perfectly connect shapes with arbitrary number of points
lets have a list of Lower and Upper points

H  H   H  H       H

   |
L        L              L0

You connect the lower with the upper plane with TRIANGLES
you start with 1st Lower point and search for most close H point for it

Then you either progress with the lower pointer or wirth the upper pointer
dependent on which new distance is shorter

I just did not yet have the temper to implement it and if either Upper or
lower plane has different number of holes, I am (xxxx)..

There is a way better option than putting in a random number - just insert
the python function code instead(but i have to do)

BTW did you notice the 7-15 version.  Did not announce it as I was afraid
of getting bugged  - but now I am more relaxed.

alacritous13, 2025-07-17:
That's smart, and should work for shapes that are arbitrarily similar to each other. That covers most cases that I would come across. Even the exaggerated example I included will probably come out fine, if slightly uncertain given starting points. To play devil advocate, how would that program work with these two point sets:

[image]

Best I can tell from your explanation, all points from one path should connect to the inner corner of the other shape.

alacritous13, 2025-08-10:
[image]
This was what I was referring to. As I've mentioned, I don't have a better solution, the math theory behind it is... subjective. That, and I intentionally came up with a formula that would break the new math formula, I don't expect to see this anytime soon.

[image]
First image is from version 2025.07.23, second image is from version 2025.07.04. 

```
from openscad import *

def CCress(h=0):#{
    c = square(4,center=True);
    c += [2,2]
    
    if h>3:#{
        # c |= square(0.1,center=True)
        c += [-4,-4]
    #}
    
    return c.mesh()[0]
#}

linear_extrude(CCress,height=6,fn=7,slices=10).show()
```

pythonscad, 2025-08-10:
I cannot clearly See w difference between the Results apart From the Viewing Angles.

alacritous13, 2025-08-10:
[image]

Here it is from the default top down view angle. Exact same code.

gsohler, 2025-08-10:
I think the issue ist that i renamed the Name of the slices From fn to
slices to conform to the openscad Syntax. I missed to document that clearly
enough.
I believe if you use slices Parameter instead of fun in the More Recent
Version, the result should Look ad expected.