pythonscad/pythonscad #82: Rendering of skin example not consistent between Guix package and local Ubuntu build

nomike, 2025-08-10:
I'm working on the build system at the moment. I've created a [package for GNU Guix](https://packages.guix.gnu.org/packages/pythonscad/0.0.0-1.e1d4903/) already.

There is an issue with a few test-cases related to the skin feature however.
My knowledge and understanding of all this 3D rendering is still quite low at the moment, so I wasn't able to figure this out so far.
It might be related to #52.

The following test cases produce a different rendering in the Guix compiled version of PythonSCAD, than they do in a version I just compiled on Ubuntu 25.04:

- rendertest_skin_edge_case-tests
- csgrendertest_skin_edge_case-tests
- rendermanifoldtest_skin_edge_case-tests
- previewtest_skin_edge_case-tests
- previewmanifoldtest_skin_edge_case-tests
- throwntogethertest_skin_edge_case-tests

I managed to simplify the issue and make the error more significant.

This code:

```openscad
skin() for (i=[0:10:400], union=false)
{
    translate([i/80,i/60,i/10])
    rotate([0,0,i/2])
    difference()
    {
        circle(5,$fn=128);
        circle(4,$fn=3);
    }
}
```

produces the following results:

**Guix:**
[image]

[Guix.stl.gz](https://github.com/user-attachments/files/21703178/guix.stl.gz)

**Ubuntu:**
[image]

[Ubuntu.stl.gz](https://github.com/user-attachments/files/21703179/Ubuntu.stl.gz)

Around the 25 mark at the y-axis you can see a bit protruding out of the slanted cylinder in the Ubuntu version, which is not there on the Guix version. If you look at the STLs you can see even more of those.

The expected-pictures of the tests are generated of course and during the creation of the Guix package, I regenrated them to make the test in Guix successful.

However, they are now failing on the Ubuntu build.

As I only have those two data points, I need other to render the same script to figure out which one is correct, but due to those weird protrusions, I currently assume the Guix version to be correct.

So could someone please render the script above and attach the STL here please?

----
I was trying a simple bisect between the Guix and the local Ubuntu build to figure out what's different.
As the PythonSCAD code-base is the same for both builds (I've used the same commit), I focused on figuring out differences in libraries.
I was able to down- or up-grade some dependent libraries in Guix and some in Ubuntu to have the same version on both sides except for these:

Library     | Ubuntu                | Guix
------------|-----------------------|--------------------
Compiler    | GCC "14.2.0" 64bit    | GCC "14.3.0" 64bit
Python      | Python 3.13.3         | Python 3.11.11
Qt          | 5.15.15               | 5.15.16
libzip      | 1.11.3                | 1.9.2
fontconfig  | 2.15.0                | 2.14.0
lib3mf      | 1.8.1                 | 2.2.0

But they all seem to not be related to 3D rendering at all. The rest of the libraries are the same (at least for them listed in `pythonscad --info`). Due to the error I got (see below) I additionally confirmed that libboost is at the same version on Guix and Ubuntu.

So at the moment I'm a bit at a loss as of what is going on.

While I was playing around with values in the design, I managed to crash the Ubuntu build once, but I was not able to repeat this.
The error output was:

```plaintext
pythonscad: /usr/include/boost/container/vector.hpp:1719: boost::container::vector<T, Allocator, Options>::reference boost::container::vector<T, Allocator, Options>::operator[](size_type) [with T = int; A = boost::container::small_vector_allocator<int, boost::container::new_allocator<void>, void>; Options = void; reference = int&; size_type = long unsigned int]: Assertion `this->m_holder.m_size > n' failed.
[1]    120421 IOT instruction (core dumped)  ./pythonscad
```
Unfortunately I do not have the core file of that. And I can't tell whether this was related in any way.

Any help on this would be appreciated, especially if someone could reproduce the issue on their system so I can rule out that it's just an issue with my local setup.

Thanks

nomike

gsohler, 2025-08-19:
> I played around with numbers and values in an attempt to simplify the code
> needed to trigger the issue.
> While I wasn't really successful, I found some other interesting behavior.
>
> I used this script:
>
> module skin_example(x) {
>     skin() for (i=[320:10:380], union=false) {
>         translate([i/80,i/60,i/5])
>         rotate([0,0,i/x])
>         difference() {
>             circle(5, $fn=128);
>             circle(4, $fn=3);
>         }
>     }
> }
> color("Tomato")
>     skin_example(2);                     // Ubuntu: filled, Guix: holecolor("Turquoise")
>     translate([12,0,0]) skin_example(1); // Ubuntu: hole, Guix: filled
>
> It renders the same shape twice, but one time with half the amount of
> z-axis rotation.
> The funny bit is, that it looks like this on the Guix compiled version of
> PythonSCAD:
> skin-issue-guix.png (view on web)
>
> But like this on the Ubuntu compiled version of OpenSCAD:
> skin-issue-ubuntu.png (view on web)
>
> I was initially trying to figure out whether the Ubuntu or the Guix
> version of skin is broken, but it looks like both are and it depends on the
> parameters you're using...
>
> If you comment out the skin() call, the rendering looks the same on both
> versions, so it's definitely an issue with the skin function.

If its only skin function, then there IS No libraries involved.
In that Case i suspect very much uninitialized variables which have
different values in guix and Ubuntu.

nomike, 2025-08-19:
I will do a general cleanup of compiler warnings for OpenSCAD and PythonSCAD. I've build both applications with -Wall and have seen plenty of warnings.

I hope that will fix this.

nomike, 2025-08-25:
I've produced some more data, which might bring us closer to an answer here.
[during debugging I produced a number of files, I will list them at the end]

I've added `printf` statements to every function in skin.cc (using CoPilot) which print the name of the function and all parameters including their values.

I built this modified version in Ubuntu and Guix, called it and redirected the output to  file.
For that I used this command line:

```shell
 ./pythonscad --o skin-issue.png ../skin-issue.scad --enable skin --render >skin-issue-3.log
```

I normaliezed the decimal separator with sed (most likely due to locale differences):

```shell
sed -i 's/\./,/g' skin-issue-*.log
```

I then opened both files in vscode and compared them.
At first there was an error message about the diff processor being killed after 500 ms for taking too long, but the error dialog had the option to temporarily lifting that constraint.

I can see differences in function skinPolygonSequence.

There is a for loop starting at line <https://github.com/pythonscad/pythonscad/blob/f0c0d2e4ffc0ee8a4bc858343deb23129a351ece/src/geometry/skin.cc#L658-L672>

In there it happens at some point, that the values of the vector components of the variables cur0, cur1, prev0, and prev1 differ.

It is usually more than by just a fraction and it never happens to the third component of those vectors, it's always the first two.

The first occurrence is this:

```diff
--- skin-issue-ubuntu.log       2025-08-25 01:45:42.986865323 +0200
+++ skin-issue-guix.log 2025-08-25 01:45:42.977090630 +0200
@@ -8040,299 +8040,299 @@
 ###check_extrusion_progression: above plane
 outputQuad: entered
 outputQuad: begin parameter list
     outputQuad: prev0 = (0,762627, 2,454577, 4,000000)
     outputQuad: prev1 = (0,817330, 2,464222, 4,000000)
     outputQuad: cur0 = (0,567614, 2,819632, 8,000000)
     outputQuad: cur1 = (0,636489, 2,844700, 8,000000)
     outputQuad: v0_progression = 1
     outputQuad: progression = 1
 outputQuad: end parameter list
 outputSingleQuad: entered
 outputSingleQuad: begin parameter list
     outputSingleQuad: prev0 = (0,762627, 2,454577, 4,000000)
     outputSingleQuad: prev1 = (0,817330, 2,464222, 4,000000)
     outputSingleQuad: cur0 = (0,567614, 2,819632, 8,000000)
     outputSingleQuad: cur1 = (0,636489, 2,844700, 8,000000)
 outputSingleQuad: end parameter list
     skinPolygonSequence: cur0 = (0,636489, 2,844700, 8,000000)
     skinPolygonSequence: prev0 = (0,817330, 2,464222, 4,000000)
     skinPolygonSequence: cur1 = (0,688686, 2,863698, 8,000000)
-    skinPolygonSequence: prev1 = (1,023565, 2,500587, 4,000000)
+    skinPolygonSequence: prev1 = (3,314178, 2,904484, 4,000000)
     skinPolygonSequence: p = 1
     skinPolygonSequence: v0 = 0
     skinPolygonSequence: v1 = 1
     skinPolygonSequence: progression = 1
```

I considered the following options:

**Floating-point math differences**

This is unlikely however, as I would expect the difference to be minuscule. But in this example it is 1,023565 versus 3,314178, which is way too big to be just a rounding issue.

**Uninitialized memory**

I don't believe that either, as then the result would not be consistent per platform.

**Operation order (unordered)**

Seems to be a candidate.

**Compiler/ABI differences**

Maybe? ¯\_(ツ)_/¯

**Eigen vectorization**

 Eigen can use different vectorization (SSE, AVX, etc.) depending on how it was compiled and the detected CPU features. This can lead to small differences in floating-point results.

But that would also more likely lead to very minuscule differences.

----
So after all, I'm a bit stuck again.
But I think I've narrowed down the area to look for significantly.

I concentrated on the first difference as the ones further on are likely only propagating from that one.

Any more ideas would be appreciated.

And here are the relevant files:

- [pythonscad-skin-issue.patch.gz](https://github.com/user-attachments/files/21959695/pythonscad-skin-issue.patch.gz)
- [skin-issue.scad.gz](https://github.com/user-attachments/files/21959696/skin-issue.scad.gz)
- [skin-issue-guix.log.gz](https://github.com/user-attachments/files/21959698/skin-issue-guix.log.gz)
- [skin-issue-ubuntu.log.gz](https://github.com/user-attachments/files/21959700/skin-issue-ubuntu.log.gz)

nomike, 2025-08-25:
I've just commented in all calls to the functions `dumpAlignmentPoints` and `dumpPolygons`, which lead to the following output files:

[skin-issue-guix-dump.log.gz](https://github.com/user-attachments/files/21959974/skin-issue-guix-dump.log.gz)
[skin-issue-ubuntu-dump.log.gz](https://github.com/user-attachments/files/21959975/skin-issue-ubuntu-dump.log.gz)

So there is also a difference visible there.

nomike, 2025-08-27:
Adding this instead of the output stream makes the rendering in Guix change as well:

```c++
#include <chrono>
#include <thread>
....
std::this_thread::sleep_for(std::chrono::milliseconds(1));
```