pythonscad/pythonscad #83: undefined reference to `vtable for RoofNode' (problem with cmake -DEXPERIMENTAL=1 -DENABLE_PYTHON=1 -DENABLE_LIBFIVE=1 ..)

DeepReef11, 2025-08-10:
Please first consider that general questions about using OpenSCAD or how to write code for specific models are better asked on the mailing list. See https://openscad.org/community.html for more details.

**Environment**
endeavour os (arch linux)

**Issue**
Missing dependency in submodule libfive.
I believe I have solved it by adding this line:
```
#include <cstdint>
```
to:
*libfive/include/libfive/tree/tree.hpp*
at line 22.

Currently building, so we will see.

DeepReef11, 2025-08-10:
I don't know why, but it seems insanely long. It has been building for at least 20 minutes on that command after adding the above change:
cmake -DEXPERIMENTAL=1 -DENABLE_PYTHON=1 -DENABLE_LIBFIVE=1 ..

DeepReef11, 2025-08-10:
The following tests FAILED:
```
        2672 - rendertest_skin_edge_case-tests (Failed)
        2678 - csgrendertest_skin_edge_case-tests (Failed)
        2684 - rendermanifoldtest_skin_edge_case-tests (Failed)
        2690 - previewtest_skin_edge_case-tests (Failed)
        2696 - previewmanifoldtest_skin_edge_case-tests (Failed)
        2702 - throwntogethertest_skin_edge_case-tests (Failed)
```

I tried installing anyway. I get this error on launching:
`pythonscad: error while loading shared libraries: libfive.so: cannot open shared object file: No such file or directory`

DeepReef11, 2025-08-11:
I found a fix. I did not ran the tests again but pythonscad lauched. I had to use:
 export LD_LIBRARY_PATH=/usr/local/lib

I don't know if that changed anything but I did go into libfive submodule and did the build instructions then `sudo make installe`.

gsohler, 2025-08-11:
> Everything seems to work fine. I haven't tried to install again but so
> far, this fixed the problem:
>
> After git submodule update --init --recursive, add:
>
> #include <cstdint>
> to:
> libfive/include/libfive/tree/tree.hpp
> at line 22
>
> then use:
> export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/libt
> (add to your .*rc file for permanent)
>
> then continue with pythonscad build instructions.
>
> If that doesn't work, building libfive (in the submodules dir) by
> following their readme instructions might solve the problem.
>
> Unfortunately, I do not know what is the actual issue that would fix this
> properly.

Thank you for Sharing your Experimence with us. These issues dont Happen
Here locally, but If you have an Idea how to improve, feel free to pr.
Another valid Option IS probably to Switch Off libfive when Not needed.