pythonscad/pythonscad #111: `memberfunction` not hot loading functions

alacritous13, 2025-09-05:
I've had limited success with loading and changing functions assigned by `memberfunction`. All tests as of this point have worked on startup, but when I add the line in an external program it has a sporadic rate of successfully assigning the member function. Additionally I've been unable to update the assigned member function after the first assignment (outside of just starting the program). It also won't update for changes to the function, I've tried flushing the cache but that doesn't work.

[image]

I've got this error message occuring, the fun part is that it's asking if I meant `TamY` instead of `TamQ`, but if I try `TamY` it'll suggest `TamQ`, niether of them were working at that time (`TamZ` was).

I've tested the program with programs being referenced with `from`, it apears that it works so far as that the first time its referenced is when its ran, and any edits after that are non-functional.

gsohler, 2025-09-05:
Yes, member functions must be defined once and correct.
But I realized, that that's a big limitation during code development, so i have improved it.

With latest commit 228aa7b34821523d0dc413d9421bae204edabefe I have successfully tested redefining member functions.

gsohler, 2025-09-05:
> I did a quick ctrl-f replace on my codebase, it crashes. I only have 12
> instances of curse, so I couldn't tell you why that is. At this point the
> crash is just hearsay.
>
> Different issue, 'memberfunction' doesn't affect operators. I've got @ (
> __matmul__) tied to the rotate function, and my testing indicates that it
> just makes the word __matmul__ do things and not the linked @ character.
> Now I'm neck deep into the forbidenfruit code messing with thing's I have
> no understanding of trying to get a foundation of what they did there in an
> attempt to add the ability to modify the comparator operators.

please send me a test code where pythonscad crashes. i am trying to
reproduce

please checkout
https://github.com/pythonscad/pythonscad/blob/master/src/python/pyfunctions.cc#L5296
this will shed some light to you

I plan to make (somehow) a setdict method on openscad type so member
functions even become easier

gsohler, 2025-09-05:
I got another solution

[image]

the patchme() is incredibly simple:

its just:

PyOpenSCADType.tp_flags &= ~Py_TPFLAGS_IMMUTABLETYPE;

Only key is where to place. it must be after openscad import.
We could hide it inside the cube() function or even rename patchme to make_mutable() ?