pythonscad/pythonscad #114: MacOS builds will not launch (missing Python directory inside app bundle)

dgatwood, 2025-09-07:
**Describe the bug**
I've made multiple attempts to get Python support to work in:

- The most recent pythonscad build (OpenSCAD-silicon-2025-04-08.dmg)
- The most recent OpenSCAD development build (as of yesterday)
- A recent OpenSCAD release build (OpenSCAD version 2025.07.20 (git 721d8bc0e))

In the OpenSCAD builds, the Python feature flag is missing from the UI.

In the PythonSCAD build above, the app will not even launch.  It says the app is damaged, and I should move it to the trash.  This is apparently because of defective symlinks.

lrwxrwxr-x  1 dg  staff      15 Apr  8 14:59 libpython3.13.a -> ../../../Python
lrwxrwxr-x  1 dg  staff      15 Apr  8 14:59 libpython3.13.dylib -> ../../../Python

First, these should presumably be to the specific files in that directory, not to the directory.  Second, the entire Python directory does not exist, which is presumably why the symlinks weren't created correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. Download any of the above builds on macOS.
2. Choose OpenSCAD > Preferences…
3. Choose Features
4. See error

**Expected behavior**
I expected a Python-parse feature flag.

**Code reproducing the issue**
N/A

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS [e.g. Windows 10, macOS 12 Monterey, Ubuntu 22.04]: macOS Sonoma 14.7.4
 - System [e.g. Intel PC 64-bit, Raspberry PI 4]: M1 Max MacBook Pro
 - PythonSCAD Version [e.g. 0.0.1, 0.1.0, etc., and where it was installed from]: OpenSCAD-silicon-2025-04-08.dmg (from https://pythonscad.org/download.php).

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `pythonscad --info` command.

**Additional context**
Add any other context about the problem here.

dgatwood, 2025-09-09:
OpenSCAD.app/Contents/Frameworks/python3.13/config-3.13-darwin/libpython3.13.a -> ../../../Python
OpenSCAD.app/Contents/Frameworks/python3.13/config-3.13-darwin/libpython3.13.dylib -> ../../../Python

If I can figure out how to get it to build from source (thus far, I'm not having much luck — cmake issues when building from homebrew, and some compile error with one of the dependencies when building them from source), I'll try to take a look at the release script.  It's on my list of things to try again when I'm not on an airplane.  😂

gsohler, 2025-09-10:
The likely source of the broken link is this one in my MAC:
/Library/Frameworks/Python.framework/Versions/3.13/lib/python3.13/config-3.13-darwin/libpython3.13.a -> ../../../Python

And this is valid link in the origin. it shows to the Python binary.

Does such a link also exist in your mac, or is that wrong ?